Ultimaker/Cura #5114: Cura 3.6.0 crashes

enginediag, 2019-01-05:
-->

**Application Version**
 3.6.0

**Platform**
 Windows 7

**Printer**
 it crashes when you try to change printer or add printer

**Steps to Reproduce**
 load up and try to change selected printer

**Actual Results**
 cura has stopped working and needs to closed

**Expected results**

**Additional Information**
 this version had been working however very slow

log

2019-01-05 01:32:39,244 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FileLogger
2019-01-05 01:32:39,264 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FirmwareUpdateChecker
2019-01-05 01:32:39,284 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FirmwareUpdater
2019-01-05 01:32:39,304 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeGzReader
2019-01-05 01:32:39,314 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeGzWriter
2019-01-05 01:32:39,324 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeProfileReader
2019-01-05 01:32:39,364 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeReader
2019-01-05 01:32:39,374 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin GCodeWriter
2019-01-05 01:32:39,394 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ImageReader
2019-01-05 01:32:39,404 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LegacyProfileReader
2019-01-05 01:32:39,424 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalContainerProvider
2019-01-05 01:32:39,434 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LocalFileOutputDevice
2019-01-05 01:32:39,454 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MachineSettingsAction
2019-01-05 01:32:39,464 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ModelChecker
2019-01-05 01:32:39,484 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MonitorStage
2019-01-05 01:32:39,504 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin PerObjectSettingsTool
2019-01-05 01:32:39,524 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin PostProcessingPlugin
2019-01-05 01:32:39,544 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin PrepareStage
2019-01-05 01:32:39,564 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin RemovableDriveOutputDevice
2019-01-05 01:32:39,594 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SimulationView
2019-01-05 01:32:39,623 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SliceInfoPlugin
2019-01-05 01:32:39,636 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SolidView
2019-01-05 01:32:39,666 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SupportEraser
2019-01-05 01:32:39,706 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin Toolbox
2019-01-05 01:32:39,726 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin CameraTool
2019-01-05 01:32:39,746 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin MirrorTool
2019-01-05 01:32:39,766 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin RotateTool
2019-01-05 01:32:39,786 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin ScaleTool
2019-01-05 01:32:39,816 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SelectionTool
2019-01-05 01:32:39,836 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin TranslateTool
2019-01-05 01:32:39,876 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UFPWriter
2019-01-05 01:32:39,896 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UltimakerMachineActions
2019-01-05 01:32:40,015 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UM3NetworkPrinting
2019-01-05 01:32:40,035 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2019-01-05 01:32:40,045 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UpdateChecker
2019-01-05 01:32:40,095 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin USBPrinting
2019-01-05 01:32:40,115 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin UserAgreement
2019-01-05 01:32:40,135 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade21to22
2019-01-05 01:32:40,155 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade22to24
2019-01-05 01:32:40,175 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade25to26
2019-01-05 01:32:40,195 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade26to27
2019-01-05 01:32:40,215 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade27to30
2019-01-05 01:32:40,235 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade30to31
2019-01-05 01:32:40,255 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade32to33
2019-01-05 01:32:40,275 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade33to34
2019-01-05 01:32:40,295 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin VersionUpgrade34to35
[... 1016 lines cut ...]
2019-01-05 01:35:56,416 - DEBUG - [MainThread] cura.Machines.MaterialManager.getAvailableMaterials [356]: Exclude materials ultimaker_tough_pla_red, generic_petg, generic_hips, ultimaker_bam, generic_tough_pla, ultimaker_tough_pla_black, ultimaker_tough_pla_green, generic_pva, ultimaker_tough_pla_white, ultimaker_pva, generic_bam for machine ultimaker2_plus
2019-01-05 01:35:56,422 - DEBUG - [MainThread] cura.Machines.MaterialManager.getAvailableMaterials [356]: Exclude materials ultimaker_tough_pla_red, generic_petg, generic_hips, ultimaker_bam, generic_tough_pla, ultimaker_tough_pla_black, ultimaker_tough_pla_green, generic_pva, ultimaker_tough_pla_white, ultimaker_pva, generic_bam for machine ultimaker2_plus
2019-01-05 01:35:56,425 - DEBUG - [MainThread] cura.Machines.MaterialManager.getAvailableMaterials [356]: Exclude materials ultimaker_tough_pla_red, generic_petg, generic_hips, ultimaker_bam, generic_tough_pla, ultimaker_tough_pla_black, ultimaker_tough_pla_green, generic_pva, ultimaker_tough_pla_white, ultimaker_pva, generic_bam for machine ultimaker2_plus
2019-01-05 01:35:56,439 - DEBUG - [MainThread] cura.Machines.MaterialManager.getAvailableMaterials [356]: Exclude materials ultimaker_tough_pla_red, generic_petg, generic_hips, ultimaker_bam, generic_tough_pla, ultimaker_tough_pla_black, ultimaker_tough_pla_green, generic_pva, ultimaker_tough_pla_white, ultimaker_pva, generic_bam for machine ultimaker2_plus
2019-01-05 01:35:56,520 - DEBUG - [MainThread] cura.Machines.MaterialManager.getAvailableMaterials [356]: Exclude materials ultimaker_tough_pla_red, generic_petg, generic_hips, ultimaker_bam, generic_tough_pla, ultimaker_tough_pla_black, ultimaker_tough_pla_green, generic_pva, ultimaker_tough_pla_white, ultimaker_pva, generic_bam for machine ultimaker2_plus
2019-01-05 01:35:56,523 - DEBUG - [MainThread] cura.Machines.MaterialManager.getAvailableMaterials [356]: Exclude materials ultimaker_tough_pla_red, generic_petg, generic_hips, ultimaker_bam, generic_tough_pla, ultimaker_tough_pla_black, ultimaker_tough_pla_green, generic_pva, ultimaker_tough_pla_white, ultimaker_pva, generic_bam for machine ultimaker2_plus
2019-01-05 01:35:56,528 - DEBUG - [MainThread] cura.Machines.MaterialManager.getAvailableMaterials [356]: Exclude materials ultimaker_tough_pla_red, generic_petg, generic_hips, ultimaker_bam, generic_tough_pla, ultimaker_tough_pla_black, ultimaker_tough_pla_green, generic_pva, ultimaker_tough_pla_white, ultimaker_pva, generic_bam for machine ultimaker2_plus
2019-01-05 01:35:56,550 - DEBUG - [MainThread] cura.Machines.MaterialManager.getAvailableMaterials [356]: Exclude materials ultimaker_tough_pla_red, generic_petg, generic_hips, ultimaker_bam, generic_tough_pla, ultimaker_tough_pla_black, ultimaker_tough_pla_green, generic_pva, ultimaker_tough_pla_white, ultimaker_pva, generic_bam for machine ultimaker2_plus
[... cut ...]
2019-01-05 01:35:58,035 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [15]: Updating CustomQualityProfilesDropDownMenuModel.
2019-01-05 01:35:58,038 - INFO - [MainThread] cura.Machines.QualityManager.getQualityChangesGroups [179]: Cannot find node for machine def [ultimaker2_plus] in QualityChanges lookup table
2019-01-05 01:35:58,041 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [50]: Updating QualityProfilesDropDownMenuModel.

nallath, 2019-01-07:
Could you try to download and run the 4.0 beta? I think your issue might already be resolved.

Vandrasc, 2019-07-01:
Hi @enginediag , I am also not able to reproduce the issue encountered by you. 

Is the issue still occurring in 4.1? Please re-open the ticket if this is still an issue for you, thank you for your feedback!

closed 2019-07-01 by Vandrasc