Ultimaker/Cura #5115: Wall missing on circular parts of print (e.g. holes)

toskium, 2019-01-05:
**Application Version**
Cura 3.6 and Cura 4.0.0 Beta

**Platform**
Windows 10

**Printer**
Hypercube Evolution (DIY Printer)
Bondtech BMG bowden style extruder

**Steps to Reproduce**
Slice the attached stl and print.

**Actual Results**
After the initial layer the printer does not deposit any material in the middle wall around one of the holes. However this only affects this specific hole in that print. The Cura gcode preview shows that there should be material being extruded.
Other prints with other curves suffer the same problem.

**Expected results**
Properly printed walls.

**Additional Information**
Attachments: 
[STL file as ZIP](https://github.com/Ultimaker/Cura/files/2729407/Duet_case_lid_cutout.zip)
[Gcode file as ZIP](https://github.com/Ultimaker/Cura/files/2729411/Duet_case_lid_cutout.zip)
[3MF file as ZIP](https://github.com/Ultimaker/Cura/files/2729416/Duet_case_lid_cutout_3MF.zip)

Cura gcode preview:
[image]

The improper hole:
[image]

Proper hole (all other holes look like this):
[image]

smartavionics, 2019-01-05:
Hello @toskium , if you look at the gcode, that wall is being printed. I think there are a couple of reasons why it is not sticking to the build plate:

1 -  you are using a negative extra prime amount so that on a travel move it retracts 5mm but then only unretracts 4.88mm. So after every retraction, it will tend to under-extrude.

2 - you are printing a small(ish) circle at 40mm/S and so unless the adhesion is good, it's probably not going to stick well.

If you use a zero or positive extra prime amount and print a little slower does it still happen?

BTW - turn on the wall printing order optimization and it will do fewer travel moves.

Hope this helps.

toskium, 2019-01-06:
Hi @smartavionics , 

thank you for your analysis. It turned out to be a combination of not enough deretraction and speed.
For whatever reason I had additional negative deretraction.

I will close that one as the problem is clearly not software related.

Thank you!
Markus

closed 2019-01-06 by toskium