Ultimaker/Cura #5116: Combing with Not in Skin doesn't work properly when using 100% infill

ethanspitz, 2019-01-05:
**Application Version**
3.6

**Platform**
Windows 10

**Printer**
[solidprint_combing_issue.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2729877/solidprint_combing_issue.curaproject.3mf.zip)

**Steps to Reproduce**
Combing on with not in skin set, use 100% infill.

**Actual Results**
Travels without retraction are occurring over the part causing stringing since Cura treats every layer like top/bottom layer.

**Expected results**
Combing with "Not in Skin" set needs to be able to identify that a layer isn't actually top/bottom even though Cura treats it like that.

**Additional Information**
Here's a screenshot that shows the issue.
[image]

My initial recommendation was going to be to fix the fact that middle layers are showing as top/bottom as that just seems wrong to me, but I saw that was raised as a bug before and closed as not fix #2670

smartavionics, 2019-01-05:
Hi @ethanspitz , I agree that those travels across the gaps look weird and I will investigate why they are not being routed over the walls/skin.

BTW - if you set the Max Comb Distance With No Retract setting to some non-zero value (I typically use 10 or 20) then those travels will use retraction and you should not get any stringing.

ethanspitz, 2019-01-05:
Great tip!! I'll give that a shot, I read about that feature yesterday, but didn't try messing with it yet.

smartavionics, 2019-01-06:
I'm now having problems reproducing this issue, can you please save the project file (which includes both the model and the settings) and zip that and attach it to this issue. Thanks.

ethanspitz, 2019-01-06:
Is that not the file I have attached in the original issue?

ethanspitz, 2019-01-06:
Ahhh my bad. Here it is. It's not exactly the settings I had when I filed this as I actually manged to corrupt my entire settings files (ran out of disk space while it was trying to save a profile...), but I managed to reproduce it identically.
[5116_Combing_issue.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/2730300/5116_Combing_issue.curaproject.3mf.zip)

ethanspitz, 2019-01-08:
Awesome! Good work! What was the bug/where is your pull request out of curiosity?

smartavionics, 2019-01-08:
Thanks. So the bug was really a flaw in the original combing implementation. The bit of code that works out whether the line from A to B crosses the boundary of the combing region was too simplistic and so although it worked OK for simple combing regions, when faced with regions that were more wiggly or that contained holes it could screw up and allow the travel path to cross directly regions that it should have followed the combing boundary.

I am not going to submit a PR until it has been tested further but I will be incorporating it into my Cura releases very soon so it will be available for people to try. You can find the current fix in the mb-fix-line-polygon-crossings branch in my CuraEngine github repo.

theWaldschrat, 2019-02-12:
Found another way to reproduce:
- Load standard profile, e.g. Draft for Ender 3
- Set horizontal expansion to -0,625 mm
- Set combing mode to "not in skin"
- Load 3Dbenchy
- Slice
- Zoom in on layers 44 to 53
- Observe unretracted travel lines.

Picture and project file added for reference.
[image]
[3DBenchy.zip](https://github.com/Ultimaker/Cura/files/2857545/3DBenchy.zip)

smartavionics, 2019-02-12:
Hello @theWaldschrat , I loaded your project into my Cura and sliced and the fix I have submitted (https://github.com/Ultimaker/CuraEngine/pull/973) has fixed the un-retracted travels. Mind you, the estimated print time and filament usage looks a bit off (that is due to something else completely unrelated to the current issue).

[image]

theWaldschrat, 2019-02-13:
Good to hear and thanks for your effort. I'll put some pressure on the PR, for what it's worth.

eskeyaar, 2019-09-19:
@smartavionics ,does your latest master release has this feature?

smartavionics, 2019-09-19:
Yes. Why, are you having a problem with it?

eskeyaar, 2019-09-19:
No just checking. Also, had issue when ironing feature is enabled. i get few travel lines over it. 
[image]
Thanks

smartavionics, 2019-09-20:
OK. Looking at that image, the travel lines do seem to be avoiding the printed model, don't they?

closed 2019-10-25 by Ghostkeeper