Ultimaker/Cura #5120: Layer View/Color Scheme color labels become unreadable

Tuggster, 2019-01-06:
**Application Version**
3.6.0

**Platform**
Windows 10 Pro, 64 bit 

**Printer**
Ultimaker 2+
[Hoadloader Clickwheel 1.zip](https://github.com/Ultimaker/Cura/files/2730108/Hoadloader.Clickwheel.1.zip)

**Steps to Reproduce**
Import model
Slice model
Switch to Layer View
Set layer view overlay to Feedrate or Layer Thickness
Select your model
Observe the color chart's labels.

**Actual Results**
[image]
The color table's labels become unreadable values

**Expected results**
The color table labels shouldn't change.
[image]

**Additional Information**
Before selecting model
[image]

After selecting model
[image]

Changing color schemes reverts the labels to their expected state.
This has happened in more than one project.

nallath, 2019-01-07:
Could you check if this is still an issue in the 4.0 beta? As far as I can see from the code, it shouldn't be an issue there.

Tuggster, 2019-01-08:
Hey, @nallath, I just downloaded the 4.0 beta, and the issue still seems to be present.

[image]
[image]
[image]