Ultimaker/Cura #5123: [3.6.0] Continuing print on top of surface that is not not on layer height boundary doesn't stick well

Yanisin, 2019-01-06:
**Application Version**
3.6.0

**Platform**
Ubuntu 16.04 LTS

**Printer**
Creality Ender 3

**Steps to Reproduce** 
Print attached STL with layer height of 0.2 mm layer height the default 0.2mm layer profile with generic PLA,  combing set to "noskin" and infill set to gyroid with distance of 5mm (not sure if this can have an effect).

**Actual Results**
The studs on the upper surface of the brick doesn't stick well. It seems like they are printed little bit above the surface. The studs can be easily snapped of ( they are the same part as the body in the STL there is no space). If the surface is at layer boundary, all seems to work well (the upper small brick) and I was unable to snap off the studs.

**Expected results**
The next layer obove the filled upper horizontal surface sticking correctly like the rest of the print.

**Additional Information**
STL that should trigger this and 3mf project: [cura_issue.zip](https://github.com/Ultimaker/Cura/files/2730392/cura_issue.zip)

[image]
[image]

Yanisin, 2019-01-06:
It seem like the layer at height 38.6 is for some reason completely missing.
[CE3_2x1x2.gcode.txt](https://github.com/Ultimaker/Cura/files/2730415/CE3_2x1x2.gcode.txt)

smartavionics, 2019-01-06:
Yes, with your profile, layer 193 is missing.

[image]

[image]

Setting the slicing tolerance to either inclusive or exclusive doesn't seem to have the same problem.