Ultimaker/Cura #5124: [3.6.0] Change bed image

4sfaloth, 2019-01-06:
This is basically a feature request, as I believe it cannot currently be done in Cura.

I think it would be extremely useful if we could:

1. Change the number of lines (and their separation) of the "chess" displayed on the print bed
2. Allow the user to upload an image of the print bed, over which the chess would be overlayed. Please see the screenshot in attachment; it shows the GUI of MatterControl adapted to a specific printer (R1+) by Robo3D. This is really the aspect of the printer's bed, which makes it very easy to see where the parts will fall on the actual. It would be awesome to be able to do the same in cura!

[image]

nallath, 2019-01-07:
The buildplate is already possible. You can just use OBJ files that have textures. This is also what the UM3 uses to display the "ultimaker" text on the buildplate.

Halestorm42, 2019-08-10:
How do you change the buildplate image? 
Cura 4.1 had an image with the Creality dragon for the Ender-3 and Creawsome had a full model of the printer. But 4.2.1 just has a blank grid with bounding box and I would like to change it to one of the others.

Ghostkeeper, 2019-08-14:
You need to set the build plate mesh to some file that has texture coordinates, such as OBJ. Then in the definition file you can set the `platform_texture` metadata to an image to use as texture. Here's an example:

https://github.com/Ultimaker/Cura/blob/fcd5a563e461972e6124e7524f4865ee84472291/resources/definitions/ultimaker_s5.def.json#L12-L13

The Creality dragon was shown by actually making a dimple in the mesh surface. That's another way to approach it.

closed 2019-08-14 by Ghostkeeper

flynetstudios, 2020-06-28:
Im trying this for the ender3, but i get like a corrputed defition files. The obj and jpg have file size limit or something special ??
Im just adding the "platform_texture" line.

``
{
    "name": "Creality Ender-3",
    "version": 2,
	"inherits": "creality_base",
    "metadata": {
        "visible": true,
        "manufacturer": "Creality3D",
        "file_formats": "text/x-gcode",
        "platform": "ender3.obj",
		"platform_texture": "enderg.jpg",
        "preferred_quality_type": "draft",
        "machine_extruder_trains":
        {
            "0": "creality_ender3_extruder_0"
        }
    },
    "overrides": {
        "machine_name": { "default_value": "Creality Ender-3" },
        "machine_width": { "default_value": 220 },
        "machine_depth": { "default_value": 220 },
        "machine_height": { "default_value": 250 },
        "machine_head_polygon": { "default_value": [
                [-1, 1],
                [-1, -1],
                [1, -1],
                [1, 1]
            ]
        },
        "machine_head_with_fans_polygon": { "default_value": [
                [-26, 34],
                [-26, -32],
                [32, -32],
                [32, 34]
            ]
        },

        "gantry_height": { "value": 25 }
    }
}``

Ghostkeeper, 2020-07-02:
No, that reference to the texture looks fine. What does Cura's log file have to say about it?

There is no limit on the file size of platform meshes or textures other than what your GPU can take. Cura's upstream repository has an automated test that prevents these files from becoming too big if you make a pull request, but that won't prevent you from doing whatever you like locally.

flynetstudios, 2020-07-02:
The obj, or stl file loads fine when i replace in the definition file, the problem just present when i try to load texture.
The log show this errors related to the creality_ender3.def.json:

2020-07-02 11:41:45,560 - DEBUG - [MainThread] UM.Settings.DefinitionContainer.getConfigurationTypeFromSerialized [265]: Could not get configuration type: Expecting ',' delimiter: line 9 column 3 (char 210)

2020-07-02 11:41:45,562 - ERROR - [MainThread] UM.Logger.logException [106]: Exception: Unable to deserialize metadata for container C:\Program Files\Ultimaker Cura 4.6\resources\definitions\creality_ender3.def.json: creality_ender3: Expecting ',' delimiter: line 9 column 3 (char 210)

2020-07-02 11:41:45,583 - ERROR - [MainThread] UM.Settings.ContainerStack.deserialize [428]: When trying to deserialize Creality Ender-3, we received an unknown container ID (creality_ender3)

2020-07-02 11:41:45,585 - ERROR - [MainThread] UM.Logger.logException [106]: Exception: Error in the format of container Creality Ender-3: When trying to deserialize Creality Ender-3, we received an unknown container ID (creality_ender3)

2020-07-02 11:41:45,599 - ERROR - [MainThread] UM.Logger.logException [110]: UM.Settings.ContainerFormatError.ContainerFormatError: When trying to deserialize Creality Ender-3, we received an unknown container ID (creality_ender3)

This is my definition file:

```
{
    "name": "Creality Ender-3",
    "version": 2,
    "inherits": "creality_base",
    "metadata": {
        "quality_definition": "creality_base",
        "visible": true,
        "platform": "ender3.obj"
		"platform_texture": "enderg.jpg"
    },
    "overrides": {
        "machine_name": { "default_value": "Creality Ender-3" },
        "machine_width": { "default_value": 235 },
        "machine_depth": { "default_value": 235 },
        "machine_height": { "default_value": 250 },
        "machine_disallowed_areas": {
            "default_value": [
              [[-117.5, 117.5], [-117.5, 108], [117.5, 108], [117.5, 117.5]],
              [[-117.5, -108], [-117.5, -117.5], [117.5, -117.5], [117.5, -108]]
        ]},
        "machine_head_with_fans_polygon": { "default_value": [
                [-26, 34],
                [-26, -32],
                [32, -32],
                [32, 34]
            ]
        },
        "machine_start_gcode": {
            "default_value": "; Ender 3 Custom Start G-code\nG92 E0 ; Reset Extruder\nG28 ; Home all axes\nG1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed\nG1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position\nG1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line\nG1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little\nG1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line\nG92 E0 ; Reset Extruder\nG1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed\nG1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish"
        },

        "gantry_height": { "value": 25 }
    }
}
```

Anyway i attach the full log file.
[cura.log](https://github.com/Ultimaker/Cura/files/4865665/cura.log)

Ghostkeeper, 2020-07-06:
That definition file is different. You're indeed missing a comma, like the error says. Please put a comma after the line defining the platform model, like this:

    "platform": "ender3.obj",

flynetstudios, 2020-07-07:
Is working now, thanks a lot. I'm trying to optimize and get the lightest model and texture detailed for share with ender3 users.
How can i tune/change the transparent shader in the model ? and elimiate the flicker in the build plate ??

[image]

Ghostkeeper, 2020-07-08:
It's not rendering the model transparently because the entire model seems to be covered with a texture. The texture is not rendered transparently, only the model itself where there are no texture coordinates. I suggest that you only put texture coordinates on the vertices that span the textured parts.

The flicker is called [z-fighting](https://en.wikipedia.org/wiki/Z_fighting). It's fixed by moving your platform model a bit down, like 0.1mm or something.

VinnyCordeiro, 2020-07-09:
It would be nice if adding a 3D model/texture for the print bed could be made by using Cura's settings menu, as it is done on PrusaSlicer/SuperSlicer, editing config text files isn't user friendly.

Ghostkeeper, 2020-07-13:
Indeed, Cura has no interface for it. It's considered that if you have the skills to design your own 3D printer or make changes to its hardware, you'll also have the skills to edit a .json file. After all, these things almost always involve customising the firmware as well. So creating custom printer definitions can only be done by manually creating those .json files.

You can still edit printers with the Printer Settings dialogue and there's a blank Custom FFF Printer available to this end, but it's limited. Flavour items such as the build plate texture are out of scope for that.

brethjjh, 2021-03-31:
how do i add it to a custom printer profile

Ghostkeeper, 2021-04-15:
@brethjjh Your custom printer profile needs to have a `platform` entry in its metadata referring to an .obj file, such as this:

https://github.com/Ultimaker/Cura/blob/4bf440b86efef17a535299b93e82b3baefcaf16f/resources/definitions/ultimaker3.def.json#L10

It needs to be .obj because that's the only format that Cura supports reading texture coordinates from.

In that .obj file you need to have the texture coordinates specified for where the texture needs to be rendered, like this:

https://github.com/Ultimaker/Cura/blob/4bf440b86efef17a535299b93e82b3baefcaf16f/resources/meshes/ultimaker3_platform.obj#L10223-L10227

And then in the metadata of the printer definition again it needs to refer to the texture to load on that model, in its `platform_texture` metadata entry:

https://github.com/Ultimaker/Cura/blob/4bf440b86efef17a535299b93e82b3baefcaf16f/resources/definitions/ultimaker3.def.json#L11