Ultimaker/Cura #5127: experimental bridging feature over support and infill

maukcc, 2019-01-07:
I love the bridging feature, but am guessing that it is actively not used over support and infill ?
We would guess that any length over "minimum bridge wall length" should be considered a bridge. Even if it is over support or infill.

smartavionics, 2019-01-07:
That's right, if a wall (or skin) has support underneath it, it won't be given the bridge treatment.

maukcc, 2019-01-08:
Then this is a bug as any length over "minimum bridge wall length" without support or infill should be considered a bridge.
So if "minimum bridge wall length" is set at 2mm, and my support or infill lines are 2.1mm apart, that should be considered as bridging from one infill/support line to the next.

smartavionics, 2019-01-08:
Interesting thought. Although, arguably, if the support lines are far apart you should be using a support interface rather than relying on the bridging to cope with the gaps.

maukcc, 2019-01-09:
As much as I love the support interface, the support interface itself also needs the bridging function.
As will the first top layer over infill.

smartavionics, 2019-01-09:
I really don't think it is the job of the bridging function to cope with sparse support. Either use denser support or campaign for support that increases density where it meets the model.

smartavionics, 2019-01-14:
Well, I don't agree. Sorry.

maukcc, 2019-01-22:
As it seems only the 2 of us have an opinion on this, and they are of the opposite nature, which is perfectly OK :), I would suggest 2 more options for bridging:
bridging over support: true/false
bridging over infill: true/false

smartavionics, 2019-01-22:
Sorry, someone else will have to do that because I don't think what you are asking for is required because...

a) infill density is under user control, you can set it to whatever you want. If you want sparse infill, use more top layers (maybe use expand skin option also).

b) support already has a mechanism for increasing the density at the top of the supported region.

Ghostkeeper, 2019-03-08:
Both infill and support have the "gradual infill" functionality which don't make it necessary to do bridging. At the steps of gradual infill the material tends to sag a bit, but that corrects itself after 2 or 3 layers.

Orum, 2019-10-29:
"Gradual infill/Infill support" does **not** solve the issue and is a different thing entirely.  Sure, it has its place, but there's no substitute for true bridging settings over gaps for materials that don't bridge well, like PETG.

Is this something that will never be added to Cura, or is its "open" status indicating that it might still happen in the future?

Ghostkeeper, 2019-10-31:
> Is this something that will never be added to Cura, or is its "open" status indicating that it might still happen in the future?

If it's open it means that we haven't shot it down yet. In the case of this one we're looking for a use case I guess. You can already change the skin settings based on your infill density because you know the infill density is the same everywhere. Some settings you can change for top skin separately from bottom skin, in case your infill density is much different from your support density. And for the case of skin over air vs. skin over support, there is an experimental overhanging wall feature that currently allows to adjust the speed (which could be expanded).

Orum, 2019-10-31:
> we're looking for a use case

How about [this](https://files.catbox.moe/tpe0q3.jpg)?  These are the same STL, one sliced in Cura, one in Prusa Slicer.  All the settings are identical _except_ for two things: ironing is off in PS (as it's not supported), and bridging settings (over infill) are enabled in PS.  Can you guess which is which?

Ghostkeeper, 2019-10-31:
So apparently you'll need to set the top skin settings to be the same as what the bridging settings are in PrusaSlicer then?

Orum, 2019-10-31:
...which gives me over or underextrusion (depending on the material), extremely slow layer times, problematic layer adhesion, and gumming on the nozzle.

agibson2, 2020-02-15:
Orum,  In case you are not following the new Beta 4.5 features...

"Bridge over low density infill. smartavionics has contributed a new setting that determines if the infill density in a location is lower than the specified infill density, skin and walls above it should be treated as bridging."

https://github.com/Ultimaker/Cura/releases/tag/v4.5-beta

Ghostkeeper, 2020-02-18:
Yup, that makes this issue fixed! Thanks for the reminder.

closed 2020-02-18 by Ghostkeeper