Ultimaker/Cura #5129: First layer above zig-zag support sometimes is parallel to support lines

vlachoudis, 2019-01-07:
Just a minor issue, I use the zig-zag support structures and sometimes the layer just above the structures is done with lines which are parallel to the support lines resulting in long bridges in between the structures.

**Application Version**
3.6

**Platform**
Windows 10

vlachoudis, 2019-01-08:
Of course, what information you would need?
I've uploaded the STL file in dropbox https://www.dropbox.com/s/e8szbebx4ocxmuj/Sample.stl?dl=0 
The cura profile https://www.dropbox.com/s/4pupnlb14cth18n/mygood.curaprofile?dl=0 
and two screenshots on layers 37 and 38 
[image]
[image]

When I load the STL I rotate it on Y (green) by -180deg
The settings are
Quality: Heigh: 0.2, Initial layer 0.3,
Shell: Wall, Thickness 0.8, Top/Bottom layers 5
Support: Zig Zag, Placement Everywhere, Density 15%
Build: Skirt, Count 2, Distance 3, Minimum length 250mm

ctmakro, 2019-01-10:
this may be the same problem as #4538

smartavionics, 2019-01-10:
> this may be the same problem as #4538

I don't think so, that was related to bridging but here we are looking at support.

I'm not an expert at support but it seems to me that if you have low support density and are not using the support interface (aka roof) then you can expect the skin and/or wall lines to fall through the gaps in the support. Isn't that what the interface is for?

gloomytree, 2020-07-07:
I have that problem, too. Any updates? Cura generated a 10 cm bridge with Zig Zag. Density had no effect, even 40% generated the bridge.

Ghostkeeper, 2020-07-09:
If you enable the extra bridge settings, there is a setting called "Bridge Skin Support Threshold". Setting that to 0 should prevent it from generating bridges, as a workaround.
However there is a different bug with that setting, too, where setting it to 0 is not working properly, so you should set it to 0.01 instead or something.

This bug where it's still bridging while the bridge is supported with support still stands, though. We didn't get around to working on this yet.

gloomytree, 2020-07-09:
@Ghostkeeper Ty, but a user might small bridges in other parts of the model. Also, I'm pretty sure the bridging settings don't affect support material.