Ultimaker/Cura #5130: Buggy pause at height plugin always leaves extruder mode absolute

smartavionics, 2019-01-07:
**Application Version**

3.6, master

**Platform**

Any

**Printer**

Any

**Steps to Reproduce**

Use pause at height plugin with the extruder mode set to relative.

**Actual Results**

The pause at height plugin unconditionally adds an M82 at the end of the pause gcode which breaks all the following gcode which is using relative extrusion.

**Expected results**

The pause at height script should leave the extruder mode the same as it was before the script was invoked.

**Additional Information**

Here's an example, relative extrusion is being used before and after the pause script which has an M82 near the end. Duh!

````
G1 X144.342 Y154.531 E0.01187
G1 X143.971 Y154.547 E0.01235
G1 X143.614 Y154.549 E0.01187
M566 X1800 Y1800
G0 F7200 X143.414 Y154.542
G0 X143.443 Y153.742
;MESH:NONMESH
G0 X137.5 Y138.5
;TIME_ELAPSED:129.472318

;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 5
M83
G10
G1 F300 Z2
G1 F9000 X190 Y190
G1 F300 Z15
M104 S190; standby temperature
M0;Do the actual pause
M109 S200; resume temperature
G1 F1500 E2
G1 F199.998 E2
G1 F1500 E-2
G1 F300 Z2
G1 F9000 X137.5 Y138.5
G11
G1 F9000
M82
G92 E0.01187
;LAYER:5
M106 S168
;MESH:Cylinder_2_20.stl
G0 X137.5 Y138.5 Z1.2
;ChangeAtZ V5.1.1: executed at Layer 5
M117 Printing... ch@L   5
M104 S190.000000 T0
M566 X1200 Y1200
;TYPE:FILL
G1 F3600 X149.999 Y150.999 E0.58791
G1 X150.069 Y151.069
M566 X1800 Y1800
G0 F7200 X152.062 Y147.75
M566 X1200 Y1200
G1 F3600 X146.75 Y153.063 E0.24988
````

smartavionics, 2019-01-07:
The G92 following the M82 is wrong also when using relative extruder mode. Those two lines need to be removed. I can see that the settings are easily accessible so it will be but a few moments work for someone to add a test for the extruder mode being relative and only output those two lines if it is not.

smartavionics, 2019-01-07:
Maybe like this:

````
  if not Application.getInstance().getGlobalContainerStack().getProperty("relative_extrusion", "value"):
    prepend_gcode += self.putValue(M = 82) + "\n"
    # reset extrude value to pre pause value
    prepend_gcode += self.putValue(G = 92, E = current_e) + "\n"
````

lion-creek, 2019-01-16:
Hi, in pause at height there is one more mistake. I'm set as shown. Once the printing continues to be pressed, the strings are pushed out and subsequently ejected. And he continues to print, but that's wrong.
[image]

;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 30
M83
G1 F4800 E-40
G1 F300 Z4
G1 F9000 X10 Y10
G1 F300 Z15
M104 S210; standby temperature
M0;Do the actual pause
M109 S210; resume temperature
G1 F4800 E40
G1 F0 E60
G1 F4800 E-40 <--------  Wrong
G1 F300 Z4
G1 F9000 X104.511 Y98.557
G1 F4800 E40
G1 F9000
M82
G92 E1648.70312
;LAYER:30

GregValiant, 2023-09-14:
This appears to have been fixed with newer versions.  Can this be closed?

smartavionics, 2023-09-15:
Sure.

closed 2023-09-15 by smartavionics