Ultimaker/Cura #5131: [3.6.0]Adaptive layer height making tree supports steeper

theboomboy, 2019-01-07:
**Application Version**
3.6.0

**Platform**
Windows 10

**Printer**
Creality CR-10

**Steps to Reproduce**
Enable adaptive layer height and tree supports

**Actual Results**
The tree support behaves differently, and in the model I found this on, it seemed like it would be weaker and less stable than the support without adaptive layers. The support went in a steeper angle, and then went straight up when it reached the x,y position of the area that needed support.

**Expected results**
The supports should look roughly the same even with the layer varience, and not go steeper than the supports without adaptive layers, which went in one diagonal line to the area that needs support

**Additional Information**
https://www.thingiverse.com/thing:2216106/files

nallath, 2019-01-08:
Why would the support need to look the same if the layer variance is different? The layer height does influence how steep of an angle will still work.

evanheckert, 2021-05-07:
It's too steep to stick. I've got it dialed way down to 20 deg branch angle, and it's still attempting over 45 deg branches. Since they're only one wall thick, they fail so easily.

Ghostkeeper, 2021-05-20:
Developers, see issue CURA-8252.

closed 2024-12-08 by github-actions[bot]