Ultimaker/Cura #5132: [3.6.0 Windows 10] Part Cooling Fan doesn't turn on during print

maxx2803, 2019-01-07:
**Application Version**
CURA 3.6.0

**Platform**
Windows 10

**Printer**
CR-10s

**Steps to Reproduce**
Simply slice a random model and start the print with the settings below

**Actual Results**
The print is carried out as it should be, except for the part cooling fan. The fan is turned off for the whole print although it should start working from layer two on.

**Expected results**
The part cooling fan should start from layer two on.

**Additional Information**
I will attach my current profile and especially the cooling settings.
[image]
[profile.zip](https://github.com/Ultimaker/Cura/files/2734034/profile.zip)

smartavionics, 2019-01-07:
Hi @maxx2803 , please attach a sample gcode file that shows this problem. Thanks.

maxx2803, 2019-01-08:
For me it is every G-Code, but here is a sample. Thank you.

[sample.zip](https://github.com/Ultimaker/Cura/files/2735570/sample.zip)

maxx2803, 2019-01-08:
Update: The issue is resolved now. Changing the G-Code flavor from Makerbot to Marlin did the job..

smartavionics, 2019-01-08:
Hi @maxx2803 , thanks for the update, sorry I had not got around to responding before.

GregValiant, 2023-09-14:
Can this be closed?

mxhffmn, 2023-09-16:
Yes. Thanks for the help!

closed 2023-09-16 by mxhffmn