Ultimaker/Cura #5140: Feature Request: Control cooling fan speed by feature type

GuiltyPixel, 2019-01-09:
While printing a part with spanning surfaces and high, thin walls with details, I found myself manually adjusting the cooling fan on the fly to achieve good results throughout the print, i.e., with my current PETG I found that 40% fan was best for top surfaces, while 50 to 100% was needed for other features based on detail size and layer times.
  I would like to be able to select fan speed modifiers based on type of feature, and size. Tiny details on a part may need cooler/slower printing or higher fan speeds, and horizontal surfaces may benefit from being allowed to 'settle' or to stay warm while the head makes the next pass as in my case. It'd be great if I could tick a box for fan control during skin/surface printing and set an alternate speed/temp/fan when the print hits a detail of a threshold size I set.

  I hope that makes sense, thanks.

nallath, 2019-01-09:
Tiny details will be pretty darn hard to detect, but i think that it's possible to change the fan per feature type.

toskium, 2019-01-14:
I would like to add, that this feature would be very useful in combination with overhangs. Currently they seem to be detected only by the bridging algorithm and further control is not possible except with the min layer time feature.

However, when printing materials like nylon or ABS, where layer adhesion suffers from constant cooling, it would be great if cooling could be tied to the line type. So you could say "0% cooling no matter the minimum layer time, but 40% cooling on overhangs only with more than n° inclination".

jonferreira, 2019-10-30:
I second this, we currently can set fan speed for Ironing for example but we also need that option for top skin I reckon

Ghostkeeper, 2019-10-31:
It's definitely possible but it'll be a bit hard to merge this into our current structure for fan speed: [Diagram](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/images/cool_fan_speed.svg)

jonferreira, 2019-10-31:
What would be the difference between implementing fan speed control for Skin layer or Ironing?

Ghostkeeper, 2019-10-31:
Not much. But there is no ironing fan speed setting either. I don't know where you found that.

jonferreira, 2019-10-31:
My bad,meant to say "bridge fan speed"!

Ghostkeeper, 2019-11-01:
Yeah, the bridge fan speed simply overrides the normal fan speed algorithm. I think that's fine though, because bridges are fairly short and usually a small fraction of the print. And you'll want them running maximum fan speed in any case.

If all structures override the normal fan speed algorithm, that means we should remove the settings:
* Maximum Fan Speed
* Initial Fan Speed
* Regular/Maximum Fan Speed Threshold
* Regular Fan Speed at Height

And it also kills the fan speed functionality that is tied to the Minimum Layer Time.

As a result, printing small features will have lots of blobs with some materials that print at higher temperatures, or the fan speed will be too high overall which would result in underextrusion. So something more clever needs to be done. Some ideas are:
* Create these 4 fan speed settings for every structure separately. That doesn't make the settings list any clearer for users though.
* Turn them into multiplicative settings, e.g. your final fan speed is going to be 80% (from the Infill Fan Speed) * 20% (the Regular Fan Speed). But that is also confusing if you don't know about one of the two.
* Do something with exception cases. Like, the Infill Fan Speed is not used any more if the layer prints in less time than the Regular/Maximum Fan Speed Threshold setting. This can result in weird border points though where hitting the Regular/Maximum Fan Speed Threshold reduces the fan speed instead of increasing it. I'm not sure how that would work since the Minimum Layer Time applies to an entire layer rather than just the infill.

GuiltyPixel, 2019-11-04:
To add to this, here's another use case, example images, and a suggested fix: When printing with low viscosity materials like ABS, selectively using the cooling fan makes it print about as well as PLA, but to avoid excessive part cooling and thermal contraction you only want it on to stabilize tiny details and overhangs which aren't 'bridges' but lack support nonetheless. Relief features like text benefit greatly from a little cooling during the perimeter wall print, but i would want the fan off (unless bridging) during the rest of the print. If you scan the gcode for lots of short movements during a perimeter wall printing operation, you can place fan speed instructions before and after that section. The existing overhang detection algorithm could probably be tweaked to look for printable but near maximum angle overhangs and do something like what the bridging fan speed setting does.

Here is an example of a feature which would have benefited from this enhancement. Text on ABS at 0.12mm layer height. Note the sagging on the top of the curved section of the 'h' and bottom inside of the 'c'. Poor adhesion and visual detail are caused by the layer not staying put. Running the fan during the segment with the unsupported layers would have done the job I think. This isn't a 'bridge' as far as Cura sees it however, so that setting doesn't work here.
[image]
Additionally, since there's no layer to 'squish' against, that section will bulge a bit, as shown, unless you reduce the extrusion multiplier for that segment too. Using my Mk. I iBalls to measure that, it looks like reducing extrusion to ~60% and hitting it with cooling would have improved that segment's visual quality and ability to adhere to and support the next layer up while sacrificing horizontal mating to the wall. As the next images shows however, it isn't attached to the wall next to it anyway :p
[image]
So basically this segment is useless except to maintain appearance and act as a support bridge for subsequent layers. It isn't mated to the layer above very well, and isn't attached to the adjacent wall at all. I'd bet that if I drag my fingernail across it, it will separate easily. If the previous perimeter line (toward inside of model) were printed at 150% width (pushing it outward a little), and this outer segment at 50% extrusion volume with part cooling enabled momentarily (now in contact with the overhang previously mentioned), we'd have a beautiful wall with a layer of approximately the correct height, attached to the material beside it and above it. The equivalent perimeter on next layer up would inherently overlap the two lines I just described, hence bonding them together. Stronger parts, better visual quality. Testing could be performed with a bit of gcode modified by hand to produce the desired effect and see if it's worth the effort and works as intended.

smartavionics, 2019-11-04:
Have you tried one of my Cura builds? There's some settings that could help with overhanging features. Might be worth trying...

[image]

You can find my builds at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

Please read the README file in there.

Ghostkeeper, 2019-11-04:
The normal Cura build has Overhanging Wall Angle and Overhanging Wall Speed, but not yet Overhanging Wall Fan Speed. I can see some value in adjusting the fan speed for overhanging walls as well.

That's different from adjusting the fan speed at every feature though.

GuiltyPixel, 2019-11-04:
@smartavionics hey thanks, i'll check them out.

MastaRob, 2020-02-24:
Was going to post a new thread, but then found this one. Am also thinking about how to make parts detailed and pretty, but also strong. So, my ideas:

- Fan speed variable by overhang angle. So you would have minimum/regular fan speed x1 at angle up to y1, and maximum fan speed x2 at angle y2 and above. Fan speed is then calculated linearly between these two angles. So e.g. I would say 20% fan at up to 30deg, and 100% at 60deg (and above). At 45deg it would then automatically give me 60% fan. This only really needs to be figured out for the outer wall lines, any lines in from the outer wall are themselves fully supported and can run at fan regular speed x1.
- User specified fan speed for infill, mentioned before and I also would like it - you really dont need fan most the time on infill, it just weakens the part.
- Re fan speed already being calculated for minimum later time, keep that setting but just go with whichever produces the highest fan speed. Being able to specify a different maximum for overhang angle compared with layer time might be nice, not essential though.
- A related but separate idea is "fan advance" - I'm running a chunky CPU cooling fan which works great, but takes about 1s to spin up. So having the fan switch on 1s before it is needed, would be really nice. Would help with small details mentioned above, bridges, and the variable fan speed for overhangs.

smartavionics, 2020-02-24:
Hello @MastaRob , almost all of your wishlist is implemented in my cura builds. See my comment from Nov 4th for a link to my builds. Please read the README.md file there for more info.

MastaRob, 2020-02-24:
Very cool, thanks I'll check it out!

MastaRob, 2020-02-25:
Wow, you really have implemented most the points mentioned! Awesome, thanks for the link. And just checking, seems you have even fixed the issue of Z hop with combing. Perfect. Dont suppose there is any way to be notified when you put out updates?

smartavionics, 2020-02-25:
Sorry, no, you will just have to check that dropbox link every now and again. I do sometimes announce the availability of new features on the UM Cura forum but not for every release.

aburgesser, 2020-05-24:
Any chance to add fan overrides for top/bottom lines smartavionics? I have been dialing down my PETG cooling to address issues with shell and walls, but I then need to add another .5 mm to my top layers to address the increased pillowing! If I could have the fan on those top layers, I could lean up my tops again.

controlflow, 2020-07-25:
It's a shame Cure still do not have out of box fan speed override per feature + fan advance :(

Sophist-UK, 2020-11-15:
See #5140 for a request for suggestion for a feature/plugin/post-processing script to vary fan speed by extrusion type (i.e. fan on for outer walls & top/bottom skin, and off for inner walls, non-skin surfaces and infill.

fvrmr, 2021-03-30:
We have created a ticket on our backlog for this.
Developers see CURA-8134

PhilBaz, 2022-06-17:
Yes, this is very much needed, especially fan speed for 'Overhanging Wall'

PhilBaz, 2022-06-17:
@MastaRob Ohh, this is good!

maruhe, 2023-01-07:
+3

mjruotolo, 2023-01-11:
+10

SteffenBlake, 2023-03-14:
Rather than explicitly just fan speed, it would be ideal if Overhangs and Bridges just had a general "Change settings" that worked the same way the "change settings on intersection" dialogue works, where we can pick and choose any settings and areas that satisfy the condition of "Is a bridge" or "is an overhang" will just have those overrides applied.

Coopski101, 2023-12-30:
Have to ditto on this. Have confirmed that on low/no cooling with abs, I get the [curling](https://forum.prusa3d.com/forum/original-prusa-i3-mk2-s-others-archive/what-to-do-against-curling-pla-overhangs/) on overhangs. If i jack up the fan speed, they look excellent, but instead of my print being in danger of getting knocked over/ugly, it doesn't have good layer adhesion because that high fan is on everywhere.

Because of the bridging settings, if it's a steep enough/specific geometry overhang, that fan speed override saves the day. Bridging settings don't save you on curves however.

Changing overhanging wall speed doesn't help because while there's "more time to cool" there's also more heat transfer into an already thin area.

freakydude, 2024-01-25:
**First of all, thank you.** 

I have been able to solve all the problems I was having with the bridge settings in Prusa/OrcaSlicer. Prusa/OrcaSlicer has problems distinguishing between strong overhangs and bridges. This is one of the reasons why settings like dynamic overhang speed, dynamic overhang fan speed and bridge fan speed are already wrong in the slicer... (https://github.com/prusa3d/PrusaSlicer/issues/12094)

In contrast, your Cura experimental bridge settings work very well for me. Unfortunately I miss the overhang feature discussed here in Cura. It doesn't need to be as bloated as in Prusa/Orca with linear functions for overhangs between 0° and 90°. 

**A rule of thumb is that large overhangs (e.g. >65°) print relatively well (at least with PLA/PETG) even without support, if you print very slowly and cool them down very well.**

Since my printer is at home and the component cooling fan is by far the loudest at 100%, I always try to keep the fan speed running as slowly as possible (and yes it has additional benefits as well). This works well if you have set the minimum layer time etc. correctly. All these settings are also available in Cura and obviously work correctly. 

For 2 things, however, it is important to cool as well as possible. At least that's my experience. Bridges and large overhangs (if you can't or don't want to print with support). With your experimental bridge settings, it's possible to override the fan speed for at least the first bridge layer. That's really enough. The rest is printed with dynamic fan speeds again, which works. (By the way, your default experimental bridging settings for a custom printer have worked out very badly for me)

**What is missing is the fan speed override discussed for wall lines with large overhangs. A simple additional parameter is sufficient for this problem. "Overhanging fan speed".**

[image]

I would also suggest changing the unit of "Overhanging Wall Speed" from % to mm/s. Because the print speed in relation makes less sense. You need to print an overhang slowly to get good results. How slow is a fixed value and has nothing to do with the print speed of the rest of your object. 

Thanks for all your work,
It would be really nice to see this feature in the near future.

GregValiant, 2024-03-22:
https://github.com/Ultimaker/Cura/blob/37d93736d8847faa442309c956b7f501089a7d28/plugins/PostProcessingPlugin/scripts/AddCoolingProfile.py#L19
"AddCoolingProfile" has been merged and is in 5.7beta.  It includes "By Layer" and "By Feature" fan control.

I'll go ahead and close this as completed.

closed 2024-03-22 by GregValiant

falkenhawk, 2024-04-26:
> "AddCoolingProfile" has been merged and is in 5.7beta. It includes "By Layer" and "By Feature" fan control.
> 
> I'll go ahead and close this as completed.

@GregValiant thanks, but the "ironing" layer - as requested in https://github.com/Ultimaker/Cura/issues/7367 - still can't be targeted with that script

GregValiant, 2024-04-26:
It cannot be "targeted" in Cura during post-process because there it is no "callout" in the gcode for ";TYPE-IRONING" .
Cura has ";TYPE:INFILL", and "TYPE:WALL-OUTER" , ";TYPE:PRIME-TOWER", etc..  The possible types are all called out in the script.
The gcode callout ";TYPE:SKIN" refers to both top and bottom skins with no mention of whether they are being ironed or not.  It's possible to know that Ironing is enabled for the print, but not which layers (or parts of layers) are being ironed.

You can use multiple instances of Advanced Cooling Control.  The first instance could be certain settings up to the "ironed" layer, the next instance would change the fan speeds for "SKINS".  That second instance could have an "Ending Layer" which would set the fan to a single speed for the rest of the print, or a third instance could set the fan speeds "By Feature" again and adjust just the SKIN speed for the rest of the print.  It is up to the user to decide which layers are being ironed.

If someone on the Cura team was to take on the project of adding the "TYPE:IRONING" line (as PrusaSlicer does) then it could be added to the fan script as the other "Types" are.  The name of this particular feature request is "Fan control by feature type" and that is what I addressed, but at this time "Ironing" is not a feature type.

I played with the script extensively when I was writing it.  The "By Feature" section is useable on large or slow models, but on smaller quick-printing models the fan speeds bounce up and down too fast for the fan to actually keep up with the changes.  You end up with more of an "average" fan speed that whatever you may have set it for.
90% of the time, "By Layer" is a better choice.