Ultimaker/Cura #5143: [3.6.0] Internal overhangs printed in mid air

AndreKR, 2019-01-09:
I'm printing [this file](https://www.thingiverse.com/download:2772014) with a layer height of 0.3 mm and a 5% grid infill. ([Cura project](https://github.com/Ultimaker/Cura/files/2742379/AI3M_Sides.3mf.zip))

The model contains some holes:

[image]

The problem is that Cura prints two layers of infill and then tries to put the bottom of the hole on top of thin air:

[image]

I kind of expected it to print (more dense) infill as some kind of internal support.

smartavionics, 2019-01-09:
> I kind of expected it to print (more dense) infill as some kind of internal support.

It doesn't treat the bottom of holes any different to any other skin region. As far as the slicer is concerned, below that hole is a solid region (after all, the model is solid there, isn't it?) The fact that you have chosen to make the solid region very porous is not taken into consideration.

AndreKR, 2019-01-09:
Sure, it is a skin region, but a very small one, smaller than the infill grid... shouldn't it print additional infill there? Aren't small skin pieces a common enough use case?

smartavionics, 2019-01-09:
You can use the expand skin setting to expand those skin regions under the holes so that they span more infill lines. That was the primary reason that setting was created (I know cos I added it to Cura). But, really you have got to expect problems when using 5% infill. BTW, you will get better support if you use gyroid infill.

AndreKR, 2019-01-14:
While the feature I was looking for (print more infill to support disconnected skin) doesn't exist, your answer kind of solves my problem, so I'm going to close this issue.

closed 2019-01-14 by AndreKR

tablatronix, 2019-10-08:
Is there a better solution to this now? Infill supports or something?

Ghostkeeper, 2019-10-10:
Infill Support will support the top side just as well as the small feature a bit lower. The only thing it really does is to not support top sides that are steeper than the overhang angle.

Cura has no real method to "aim" infill at protruding pieces except Gradual Infill and the Cubic Subdivision pattern. Gradual Infill also supports the top side just as much as the rest but will make the denser infill a bit lower for your internal cavity. Cubic Subdivision places denser infill all around the protrusion but in this case will have very little effect since the object itself is fairly thin already.

tablatronix, 2020-09-02:
I still have yet to find a good solution to allow holes, bosses and features to be excluded from infill, or have a denser infill, so they are not using the top skin for strength.

Could I use mesh overlap overrides?

[image]

interestingly enough, I have managed the opposite, no infill... useless but interesting. ( this is infill support <90º)
[image]

Ghostkeeper, 2020-09-03:
Did you try the (relatively) new Skin Edge Support Thickness setting?

tablatronix, 2020-09-03:
I will try it, I think I have not installed the update yet

printingotb, 2021-01-08:
You can use a denser infill with a couple of "gradual infill steps" enabled. That way it will halve the density for each step you add. So if you set it 20% and two steps, it will print it at 20% only near the top layers and then reduce it first to 10% and then to 5%. Use the  "Gradual infill step height" setting to determine how far below the supported layers it does it.

tablatronix, 2021-01-08:
Just an update I have been using Skin Edge Support constantly now, Good tip for the infil steps, I tried it previously but could not get the gradation right, but did not really understand it at the time.