Ultimaker/Cura #5144: [question] Finding containers

fieldOfView, 2019-01-09:
I'm trying to find the number of quality profiles available for a configuration of an extruder. The following code works:
```
        search_criteria = {
            "type": "quality",
            "material": fallback_material
        }

        if global_stack.getMetaDataEntry("has_machine_materials"):
            search_criteria["definition"] = global_stack.definition.id
        else:
            search_criteria["definition"] = "fdmprinter"
        if global_stack.getMetaDataEntry("has_variants"):
            search_criteria["variant"] = extruder_stack.variant.name

        containers = container_registry.findInstanceContainers(**search_criteria)
```
Back in the time, there used to be a lot of these findInstanceContainers queries, but they have all gone. That makes me feel I am doing things "the old way", but I can't really find the new way to do it. 

I am specifically NOT looking for a quality group because I want to know what is available for a specific extruder instead of for the combinations of all active stacks (global + extruders). So I think QualityManager.getQualityGroups is not what I am looking for (and that looks very complex for what I want to do).

As I said, the above code works just fine, with the exception of it not respecting build plate variants (...), but I am wondering if there is now a better way.

nallath, 2019-01-10:
I guess that looking at a single extruder just isn't done by us anymore (since we generally don't care what's there for one extruder, we care about the groups).

That being said, it's faster to use the findContainersMetadata function instead of the findInstanceContainers, since we only load the metadata on boot. If you use the findInstanceContainer we actually initialize the results (eg; So say you get 5 hits, you will get 5 instantiated instance containers, so if they werent loaded before you called it, they are now).

As you're only looking for a number, the number of metadata entries should do the trick just fine.

fieldOfView, 2019-01-10:
Thanks. The reason I'm looking at a single extruder is that I want to show the user what extruder is the cause of there not being any profiles:

[image]

Question answered, closing the issue.

closed 2019-01-10 by fieldOfView