Ultimaker/Cura #5145: Sliced object height is scalled to 15mm

paulkoug, 2019-01-09:
**Application Version**
Cura 3.6.0

**Platform**
Windows 10 64bit

**Printer**
Custom printer

**Steps to Reproduce**
Slicing any object bellow 15mm height
Upload to Octoprint
Print

**Actual Results**
Any object bellow 15mm height, its printed at 15mm height, with bigger gaps between layers and underextrusion. The layer number remains the same. All other axes are fine.

**Expected results**
Normal size printed

**Additional Information**
Slicing the same object with other slicers seem to work just fine. I uninstalled all plugins from octoprint just in case but the problem didn't go away. I tried with more than 2 objects. I ininstalled and installed again Cura.

smartavionics, 2019-01-09:
Hello @paulkoug , please zip up some gcode that exhibits this problem and attach to this issue. Thanks.

paulkoug, 2019-01-10:
> Hello @paulkoug , please zip up some gcode that exhibits this problem and attach to this issue. Thanks.

I uploaded it at the original post. (Thumb.zip)

smartavionics, 2019-01-10:
Thanks for the files. Well, the gcode looks right, I do not know why it gets scaled to 15mm. What happens when you print the same gcode using another host?

[image]