Ultimaker/Cura #5153: Arrange All Models when using Support Blocker arranges the support blocker as a separate model

pfjason, 2019-01-12:
**Application Version**

3.6.0
**Platform**
Windows 10
**Printer**

Any
**Steps to Reproduce**

Add a block support block to a model that is not centered on the build plate
**Actual Results**
The Block Support block moves in relationship to the model that it was created on

**Expected results**
The "block support" block should keep it's position in relationship to the model 

**Additional Information**
Before
[image]
After
[image]

rburema, 2019-01-15:
Devs: Internal ticket CURA-6120 was made due to this issue.

Ellecross, 2019-02-15:
Hey @rburema the CURA-6120 is fixed. :D Good work! I just checked it now. 
@pfjason a fix was applied and tested for this issue. This fix will be available in the next Cura 4.0. Thank you for reporting it! Wheeeeee!

rburema, 2019-02-15:
Should be in 4.0-BETA-2 as well (_should_ be later this month I think) if you want to check and don't want to wait for 4.0 final release (which probably won't be a long wait counting from beta-2, but still).

closed 2019-02-15 by rburema