Ultimaker/Cura #5154: Cura 3.6 missing Browse Packages in Marketplace?

mikewo, 2019-01-12:
**Application Version**

3.6.0

**Platform**

Windows 10

**Printer**

N/A

**Steps to Reproduce**

Installed Cura after downloading, no issues with the install. I included the Custom FFF Printer because my printer is not one of the officially supported ones. I was going to download the extension that supported it (Dremmel IdeaBuilder). Clicking on Marketplace it pulls up the dialog, but I do not see a Browse Packages under the Marketplace menu, on the dialog, or anywhere else.

**Actual Results**

Can't find the Marketplace browse feature at all

**Expected results**

**Additional Information**

fieldOfView, 2019-01-12:
What exactly do you think you are missing with "Browse packages"? All plugins that are available through the Marketplace for Cura 3.6 are listed in the Marketplace dialog.

Unfortunately the Dremel plugin has become unavailable for Cura 3.6, when it was published for Cura 4. This is an architectural oversight in the Marketplace backend.

@diegopradogesto, see, this is a problem.

diegopradogesto, 2019-01-12:
> @diegopradogesto, see, this is a problem.

Yep, I know and we are currently working on fixing it. Hopefully next week we will have a solution. Then we'll send an email to all the contributors indicating how you need to proceed.

I'll link this issue to our internal ticket as a reference.

ChrisTerBeke, 2019-01-12:
It's actually not a problem in the Marketplace back-end, which fully supports multiple Cura versions per package. It's just that plugin.json does not, which is something that will be fixed soon as Diego mentioned.

fieldOfView, 2019-01-12:
> It's actually not a problem in the Marketplace back-end, which fully supports multiple Cura versions per package.

But that _is_ a problem. It should support multiple versions of single packages. I don't want my plugins for 3.6 to become unavailable when I publish a new version for 4.0; I also don't want to keep updating and testing my plugins for 3.6 when I publish newer versions for new versions of Cura.  I just want the old version to stay available for 3.6.

Sorry for taking this quite off-topic, but take the OctoPrint plugin as an example. For 4.0, I need to make significant changes (eg to the user interface) that do not work at all with 3.6. So I cannot publish one curapackage that works with both 3.6 and 4.0 even if I were able to somehow mark compatibility with multiple SDK versions in plugin.json. I need the previous version of the plugin to stay available for people using previous versions of Cura.

ChrisTerBeke, 2019-01-12:
The back-end is modeled after how both Google Play and the Apple App Store work, both only allowing a single version of an app to be published at at time with support for multiple OS versions needing to be done in the app itself. Another reason we want this is because of the material package supports, where we want the materials to be updated by the vendors and usable in multiple versions of Cura. So we're very likely not going to change that system. By supporting multiple Cura versions in plugin.json, we can at least cover a lot of use cases already. But you're right that UI work is not easy to support in multiple versions (apart from having separate QML files and loading the correct one depending on the Cura version).

fieldOfView, 2019-01-12:
> The back-end is modeled after how Google Play and the Apple App Store work

I'm sorry to say, but this is incredibly naive. Neither iOS not Android have breaking changes in just about every update (even between betas!). Both OSes have backwards compatibility with older apps that Cura can only dream of. You have a lot of users that wait before upgrading to newer versions of Cura or even run multiple versions of Cura side-by-side. 

Note that it is not just UI work. Hardly a release goes by that I don't have to make changes to how my plugins work (especially for OctoPrintPlugin), so much so that OctoPrintPlugin has to do its own version-checking or Cura will crash after updating.

ChrisTerBeke, 2019-01-12:
What can I say? There's more factors that went into this decision. All I can do is go back to the group of people involved in this and discuss if we can or want to change this behaviour. You're totally right about backwards compatibility.

fieldOfView, 2019-01-13:
See also the forum:
https://community.ultimaker.com/topic/26176-plugins-within-cura-are-missing/

mikewo, 2019-01-13:
As an end user I would have to say that having a package just disappear because it was published to be ready for a BETA version of the program is a very poor user experience. I only mentioned the missing browse because instructions I've seen include using it. It's not clear in the UI that the full marketplace is shown in the dialog, especially since it contains the "Featured" label at the top. I do see the color delineation between featured and community plug-ins, but I had interpreted it to be that these were just the featured 1st party and community plugins, not ALL of them. That could have been because I was looking for the "Browse" already which led me to believe that there were more than what was showing; however, I'd almost say that many people familiar with marketplaces would see the word featured and assume that meant there were more you weren't seeing. Perhaps a better word could be used rather than featured. 

I'll install the beta 4.0 to get at the dremel plug in.

diegopradogesto, 2019-01-13:
@mikewo , as a workaround, you can also install the Dremel plugin for 3.6 by downloading it from the website of our contributor: https://github.com/timmehtimmeh/Cura-Dremel-3D20-Plugin/releases

closed 2019-10-09 by mikewo