Ultimaker/Cura #5155: Cura 3.6 disables heating while printing dual extruder

szerwi, 2019-01-12:
**Application Version**
3.6

**Platform**
Windows 10 64-bit

**Printer**
Custom FFF Printer (GeeeTech A10M)
[Project.zip](https://github.com/Ultimaker/Cura/files/2752336/Project.zip)

**Steps to Reproduce**
The problem is happening only on dual-extrusion models

**Actual Results**
On most dual-extrusion models Cura disables hotend heating at about ~80% of printing. I have tested 5 different models and it disables heating on 4 of them. It happens on last few layers.
[image]

**Expected results**
It should heat till the end of the print

smartavionics, 2019-01-12:
I sliced the cow model and it sets the T1 heater to zero after the last layer that uses T1 has been printed. I am not familiar with using 2 extruders so tell me why does T1 have to stay hot if it is not going to be used again in the print? Thanks.

szerwi, 2019-01-12:
@smartavionics I have dual extruder and single nozzle printer so it has only one heater. But I think there is no way in Cura to set that it is single nozzle.

smartavionics, 2019-01-12:
Thanks, I now understand the problem. I think you are right that Cura doesn't understand multiple tools with a single heater. Sorry, I can't really help but maybe someone else can suggest how this could be fixed.

fieldOfView, 2019-01-12:
You could use a Search & Replace postprocessing script to remove that ```M104 T1 S0```

szerwi, 2019-01-12:
@fieldOfView Thank you, it is working. But I think Cura team should add option that the printer is dual-extruder and single-nozzle.

fieldOfView, 2019-01-12:
> But I would be grateful if Cura team would add option that the printer is dual-extruder and single-nozzle.

Fixed that for you.

Note that most of the Cura team is paid by Ultimaker to support Ultimaker printers. There are no Ultimaker printers that come with a dual-extruder single-nozzle setup. So I would not hold my breath if I were you, especially since a workaround is available.

nallath, 2019-01-14:
FieldofView is right. There are no (fulltime) non Ultimaker developers working on Cura (the only non-ultimaker employees working on Cura are community contributors such as fieldOfView). The concept "Who pays, decides" is also true in this case. 
As there are still piles and piles of things to fix that do benefit Ultimakers, there is little chance of us picking it up. 
Your best bet would be to convince the GeeeTech devs to contribute back to Cura.

smartavionics, 2019-01-14:
How about this: Aldo does the python bit and I will do whatever is needed in the engine?

smartavionics, 2019-01-18:
I have posted a question on the Duet forum to see what people expect in this regard.

fieldOfView, 2019-01-18:
Also a list of what would need to change in the gcode output would also be handy. Is the issue mentioned in the opening post the only thing standing in the way of happy single-nozzle-dual-extrusion prints, or are there other hoops to jump through that could be improved upon?

Remember, I don't have a single-nozzle-dual-extrusion printer, so I genuinely don't know what needs to be done.

smartavionics, 2019-01-18:
> Remember, I don't have a single-nozzle-dual-extrusion printer, so I genuinely don't know what needs to be done.

Neither do I so that's why I posted on the Duet forum as I know there are people there that do.

smartavionics, 2019-01-18:
Hi @fieldOfView , this is the reply I got on the Duet forum from the main developer of RepRap firmware.

>The best option is probably to let RepRapFirmware do the tool temperature management when tool changes occur. RepRapFirmware supports separate active and standby temperatures to handle the case of tools with independent heaters, where you want the standby tool to be maintained at a lower temperature.

>In machines with separate heaters for each tool, setting a tool's temperature to 0 is OK for an IDEX-style machine, but it is probably a bad idea fo machines with dual nozzles on a single print head. A nozzle that is heated to standby temperature may skate over small protrusions in the print, whereas a cold nozzle is likely to ram into them.

>Don't forget that you can have machines with both mixing nozzles and a separate nozzle for support material or high-speed infill. We have a printer like this. So assuming that the machine has either a mixing nozzle or separate heaters for each tool doesn't cover the more general case.

>IMO the real problem is that other 3D printer firmwares don't handle multiple tools in a flexible way or abstract the details of a tool (e.g. heaters used) away in the way that RepRapFirmware does. Slicers shouldn't need to know about tool heaters, only about tools. it's daft to have to configure both the printer firmware and the slicer to know what the mapping between heaters and tools is.

> The logical solution to your question is that if you want to turn off tool heaters early, the slicer should send a "finished using this tool" command to the printer, and the printer firmware should decide what if anything to do about it.

The thread is here https://forum.duet3d.com/topic/8663/info-request-single-heater-with-multiple-extruders

So, as usual, my simplistic idea of what's required is not really a good general solution.

smartavionics, 2019-01-20:
See also https://community.ultimaker.com/topic/25233-auto-inserted-m104-t0-s0-during-print-with-dual-extrusion/

Ah-ah-ron, 2019-01-31:
I also had the problem with printing from the second extruder of a GeeeTech A10M where the nozzle would not heat up.  I came up with a fix which seems to be supported by the printer firmware.  You don't have to modify the g-code.  On the printer control panel, simply go to Prepare=>Preheat PLA.  Then go to the top of the menu structure again and Control=>Temperature=>Nozzle and specify the temperature you want.  Once the printer nozzle reaches the desired temperature, then download the model from the PC using the USB cable.

hussainsail2002, 2019-03-29:
Hi,

I seem to be having the same issue. 

Using a Chimera Plus for dual extrusion.
The nozzle temp of the idle nozzle drops to 200 or 205 even though I have set the standby temp as 210

Also it does not wait for the printing nozzle to heat up to the printing temperature from the standby temp.

This has been leading to nozzle clog issues.

Any help is much appreciated.

(Using Chimera plus with dual direct drive. firmware is marlin 2.0)

fursphere, 2019-05-24:
Not sure if this is still a problem or not, but if someone finds this post...   

I just made sure to set both extruder temps (even though there is only 1 heater) to the same temp in the slicer.  I don't know if this is a Cura 4.0 fix, or a slicer work around.  But I don't have the issue.

smartavionics, 2019-05-24:
Hello @fursphere , this is a current topic, see https://github.com/Ultimaker/Cura/issues/5770 for the latest developments.

Ghostkeeper, 2019-05-28:
> I just made sure to set both extruder temps (even though there is only 1 heater) to the same temp in the slicer. I don't know if this is a Cura 4.0 fix, or a slicer work around. But I don't have the issue.

This is a workaround. Since Cura will assume that there are 2 heaters, it'd normally output temperature changes for both heaters which in your case lead to the same heater which would turn off the heater after one of the extruders becomes unused for a while. If all temperatures are the same (in particular the stand-by temperature), it no longer needs to send temperature changes so it won't suddenly cool down the inactive nozzle.

andymozza, 2019-05-28:
Just to echo @smartavionics comment above, I have been testing an idea he had and respective changes to code on a three tool  (100% black, 100% white and 50/50 mix) - ( shared heater) printer and have now gone through 11 prints where various tools are no longer used for the rest of the print, but the standby temp and heater offs are no longer an issue, print completes correctly. @Ghostkeeper unfamiliar with the process to request this gets included in main cura at some point ? Can we re-open or re-label #5770 so that people can find it for comment more easily ?

Ghostkeeper, 2019-05-29:
Yeah, well, actually I think that this one is a better candidate: https://github.com/Ultimaker/Cura/issues/4204 More to the point, at least.

In fact, I think I'll close this one as a duplicate of #4204. Please continue discussion there (unless you disagree).

closed 2019-05-29 by Ghostkeeper