Ultimaker/Cura #5156: Unnecessary supports

dagnabitboy, 2019-01-12:
I created a simple test project to illustrate an issue I'm having with a more complex print.  Basically, a lower part of the print has openings with 45 deg angles and do not need support (when angle is set to > 45).  However if there is an opening higher up in the model that needs support, the supports created by cura travel all the way through the model and fill in the open space below that does not need support.  Sorry if my description isn't well written, but the pictures I attach show the issue well.  

Another strange thing is that the unnecessary supports (in the angled opening) aren't even attached to the model!  In the actual model I wanted to print the were about 50 or so opening like this, so it makes quite a mess because the unneeded supports are attached to the lower part of the openings and still need to be removed!  I played with all the settings I could find but could not get around this issue.  Support blocker has no effect.  Any suggestions (or maybe this is a bug) are most welcome.  Thanks very much in advance!!

Cura 3.6.0
Win 10 pro
Creality CR-10s
Model created in Tinkercad

[image]
[image]
[image]
[SupportTest-1.zip](https://github.com/Ultimaker/Cura/files/2752447/SupportTest-1.zip)

discharged-spider, 2019-01-13:
Hi! Looks like duplicate of I think this is a duplicate of [#4583](https://github.com/Ultimaker/Cura/issues/4583).
P.S. It's really strange to me that this bug hasn't been fixed for so long.

dagnabitboy, 2019-01-13:
Yes, it looks like the same issue.  Tree supports seem to avoid unnecessary areas, but they always start at the build plate so there is a massive amount of extra extrusion going on.  Hopefully my post will act as a 'bump'!!

rburema, 2019-01-15:
Could you try the 4.0-BETA-1? This bug was largely fixed (though in the case of this model it still tries to support the part of the upper support that's outside of the X/Y bounds of the model, leaving two thin 'walls' behind).

Ghostkeeper, 2019-01-15:
Yup, duplicate of #4583

closed 2019-01-15 by Ghostkeeper

dagnabitboy, 2019-01-15:
Thank you Ghostkeeper and rburema.  I will give 4.0 BETA a try.  That's good to know about tree supports.  Always impressed at the responsiveness of folks in this forum!!

Willowcookie, 2019-02-06:
Got this issue too, its really annoying. FYI this bug does not exist in cura versions 3.2 and below. But still not fixed.
[image]

fieldOfView, 2019-02-06:
Have you tried the Cura 4 beta?