Ultimaker/Cura #5159: Artifacts in/on Part

tukbuk23, 2019-01-13:
**Application Version**
4.0.0. Also occurs in 3.6.0.

**Platform**
Windows 10 x64.

**Printer**
Monoprice Select Mini v1.
[MSMV1_keycap.3mf.zip](https://github.com/Ultimaker/Cura/files/2753372/MSMV1_keycap.3mf.zip)

**Steps to Reproduce**
1. Open attached project file.
2. Slice and open layer view (preview).

**Actual Results**
Artifacts appear on the part walls even though part is smooth. 

**Expected results**
Part walls should consist of smooth lines.

**Additional Information**
See attached screenshot.
[image]

smartavionics, 2019-01-13:
Turn off coasting.

tukbuk23, 2019-01-13:
That fixed it. Many thanks!

closed 2019-01-13 by tukbuk23