Ultimaker/Cura #5162: [4.0.0 BETA] 100% infill not fully solid when top/bottom layers overridden

frank26080115, 2019-01-14:
**Application Version**
4.0.0 beta

**Platform**
Windows 10

**Steps to Reproduce**
Set infill to 100%, doesn't matter which pattern
Set (override) Top Layers to 3
Set (override) Bottom Layers to 3

**Actual Results**
Slicing results in 3 bottom layers, 3 top layers, but everything in between is not fully infilled, it looks like about 50% instead of 100%

Feedrate preview shows 3 layers at the bottom slow, 3 layers at the top slow, infill done very fast

**Expected results**
3 bottom layers slow, 3 top layers slow, everything inbetween totally solid done very fast

**Additional Information**
Attaching 3mf file
[rodholder_infillbugreport.zip](https://github.com/Ultimaker/Cura/files/2753892/rodholder_infillbugreport.zip)

If I don't override the top and bottom layer counts, then the print is immensely slowed down because the infill speed setting isn't respected

frank26080115, 2019-01-14:
work around found: the infill pattern needs to be set to "lines"

would still like it to be more intuitive where 100% means solid, maybe hide the option for pattern or override the pattern to lines whenever 100% is used

rburema, 2019-01-15:
Possible duplicate of #5157 ?

frank26080115, 2019-01-15:
Absolutely not

My report is pointing out the fact that currently, setting infill 100% will cause Cura to automatically set bottom layers count to 99999, which means the print will take a much longer time because the infill print speed is not being used, the bottom layer speed is being used. Manually overriding top and  bottom layer count will cause unexpected behaviour with respect to infill placement for most infill patterns

It seems like infill pattern "line" and "concentric" are the only ones that work in all scenarios. I want other infill patterns to work in this scenario as well (scenario meaning, 100% infill, top and bottom layer count set to a non-zero non-infinite number manually)

I understand that it makes no sense for there to be "100% triangular infill" due to the ways the lines will overlap. So I am suggesting that if 100% is used, the infill pattern defaults to "lines" automatically (but the drop down should not be changed so the user doesn't have to change it back to triangles later)

I hope I have provided enough information to proceed with development of a solution that is intuitive to all users.

rburema, 2019-01-16:
We've discussed this issue, and came to the conclusion that it would be best to warn the user if an incompatible combination such as 'triangle infill'/'100%' is detected. However, we can't currently warn on combinations, or enumerations, but only on min/max (which makes the field orange, or even red). Therefore, the user would currently get no feedback on _why_ 100% infill gives a warning in specific cicrumstances (which could be because they selected and incompatible infill type).

So a full resulution to this issue will have to wait until the warnings system can be improved.

stopanko, 2023-05-09:
@frank26080115 
How Cura doing it(In a bad way).

Let's say you have 3 Bottom Layers.
Infill 15%.

- You set 100
- Curra change Bottom layers to 999999. S they are printed not like `Infill`

Hot to fix if you have `100%`. But still see infill in a slicer.

- Set 100% infill.
- Go to Top/Bottom
- Near `Bottom Layers` and `Initiall Bottom Layers` you can see **Function Button**
- I assume this button meant to reset value to he **Logic** that needed.
- Hit it.
- Now with 100% Infill you should have 999999 Bottom layers. With less -> you should see Infill in place