Ultimaker/Cura #5164: [Cura 4.0.0 Beta] Gcode Preview does not show bridge with multiple layers

toskium, 2019-01-14:
**Application Version**
Cura 4.0.0 Beta

**Platform**
Windows 10 Pro

**Steps to Reproduce**
Create a simple bridge test and set the parameter "Bridge Has Multiple Layers" to true.

**Actual Results**
Cura gcode preview only shows the first layer of the bridge and ignores the multiple layers option.

**Expected results**
The Cura gcode preview should show the multiple layers of the bridge.

smartavionics, 2019-01-14:
Hi @toskium, please provide a screenshot showing the problem and also save the project file, zip it and attach to this issue so it can be investigated. Without that info, we can't do much. Thanks.

toskium, 2019-01-15:
Hi @smartavionics,

I found my mistake by analyzing the gcode myself. Cura isn't responsible for this.

Cheers!

closed 2019-01-15 by toskium