Ultimaker/Cura #5171: ("Pre") End Gcode bug/glitch causing major problems

alecfs24, 2019-01-15:
So basically, at the end of every print, ALL AXIS ZERO OUT. Thats not good in several ways. In doing that Its caused my printer to nearly break itself and definitely break some prints, the larger they are, the worse the damage.

Cura beta 4.0.0

Monoprice Maker Select v2

There are no steps that I can say, but its done it every time now, and once it finishes zeroing out, it then proceeds to do the end gcode. Previously, however, with 3.5.1/3.6.0 It has just "zoned out" on the x-axis and just kept going past max, grinding the belt, so I had to cut power. As I said, I don't know how to give steps on reproducing it, because it just happens, and all axis move at like 15mm/s home at the same time, not one at a time, then the end gcode... hope that helps. I would include a gcode file, but I don't know of a command that zero's the axis's all at once, so I fail to see what purpose one would serve.

smartavionics, 2019-01-15:
Please zip up the gcode that has this problem and attach to this issue. Thanks.

Ghostkeeper, 2019-01-15:
> I don't know of a command that zero's the axis's all at once,

What about `G28` homing? Or maybe `G0 X0 Y0 Z0`?

closed 2019-02-23 by alecfs24