Ultimaker/Cura #5173: [3.6.0] Unexpected directory created

LeMondeaLambert, 2019-01-15:
**Application Version**
3.6.0

**Platform**
Windows 10

**Printer**
Creality CR-10

**Steps to Reproduce**
Double click on a .stl file in a  windows explorer

**Actual Results**
A directory named "%HOMEDRIVE%%HOMEPATH%" is created beside the .stl clicked file.
Inside it a directory called "AppData" is created.
Inside it a directory called "Roaming" is created.
Inside it a directory called "cura" is created.
Inside it  there are 2 files, called "stderr.log" and "stdout.log"

This occurs in every directory where I have double clicked an .stl file !

**Expected results**
This should have occured in my case in:   "C://Users/Patrick/AppData/Roaming/cura"
right ?

**Additional Information**
[image]

LeMondeaLambert, 2019-01-15:
> Can't reproduce.
>
> Maybe the %HOMEDRIVE% and %HOMEPATH% environment variables are not defined on your system? Which is weird, because they should always be defined on Windows.
>
> These variables are not even used by Cura itself but by Python on which Cura runs to get at your application data folder.

It is weird

If I look via the command prompt, it looks good,
when I type set, I get all my environment variables,  including those:

HOME=C:\Users\Patrick
HOMEDRIVE=C:
HOMEPATH=\Users\Patrick

But when I look via Control Panel, they don't appear in "Environment Variables"
HOME appears, but not the others ...
And in fact it seems there are very few (8) "User variables for Patrick" set !!!!  

My PC is pretty old, 2013 I think; was running Windows 8 then Win10 and I am a heavy user,
I am a retired Windows C++ developer, working 10 hours a day on my PC,
doing lot of things...  my PC is probably corrupted, but seems to run ok.

It seems that I lost many of my environment variables !

So I tried the following:
Close Cura
Set those two variables back,
Delete those %HOMEDRIVE%%HOMEPATH% directories
Double click on a .stl
and Oup's .. %HOMEDRIVE%%HOMEPATH% directory is back

Maybe a reboot is needed ...

YES !  now it works !
No more unexpected directory.
 
Thank's to guide me in the good direction !
Cheers
Patrick

closed 2019-01-15 by LeMondeaLambert

Ghostkeeper, 2019-01-15:
I'm not currently on Windows, but I also don't recall `HOMEDRIVE` and `HOMEPATH` ever appearing in my environment variables dialogue either, but Cura doesn't create any directories in random folders for me.

Anyway, I'm glad that a reboot worked (and you shared your solution). That should serve as a workaround for people to find in the future.