Ultimaker/Cura #5174: [master/4.0] material weight should round up to a non-zero value

smartavionics, 2019-01-15:
**Application Version**

master/4.0

**Platform**

Any?

**Printer**

Any?

**Steps to Reproduce**

Slice a weeny model.

**Actual Results**

Zero grams of filament is required to print this model.

**Expected results**

Required filament weight should be non-zero.

**Additional Information**

[image]

Ghostkeeper, 2019-01-15:
So I sliced a 7mm cube of PLA, with brim, got a g-code saying that it used 370mm³ of material. According to [this calculation](https://www.wolframalpha.com/input/?i=370mm%5E3+*+1.32g%2Fcm%5E3) this should result in 0.49g of PLA. This gets rounded down to 0g, so far so good.

Then I sliced a 8mm cube. This resulted in 482mm³ of material, which should result in [0.64g](https://www.wolframalpha.com/input/?i=482mm%5E3+*+1.32g%2Fcm%5E3) of material. So this gets rounded up and the interface now displays "1g".

The rounding seems to be working fine, to me. Do you think that there should be a bias to always round up or something?

smartavionics, 2019-01-15:
I would expect (hope) that amounts would round up to the nearest 0.5g (for small amounts and round up to the nearest gm for larger amounts.
Apropos your example, it seems inconsistent to me that 0.49 is rounded down to zero and 0.64 is rounded up to 1. Why isn't 0.64 rounded down to 0.5?

Ghostkeeper, 2019-01-15:
Because it rounds to the nearest 1g, not the nearest 0.5g.

smartavionics, 2019-01-15:
> Because it rounds to the nearest 1g, not the nearest 0.5g.

Indeed.

I still think displaying zero for the filament weight is unhelpful. Still, life's too short to worry about this when there's so much else that needs fixing. At least we had the conversation, bye!

closed 2019-01-15 by smartavionics