Ultimaker/Cura #5178: 3.6.0 Screen flicker

tonymaglioni, 2019-01-15:
**Application Version**
3.6.0

**Platform**
Windows 7

**Printer**
Ender 3

**Steps to Reproduce**
I get an annoying screen flicker when using Cura. Anytime that the mouse is moved onto certain places on the screen (like when you you are clicking on drop down menus etc.) I am just using onboard graphics and wondered if my graphics aren't good enough. However, I had Cura installed on a slower computer also with onboard graphics and got no flicker. I have uninstalled and reinstalled, but it remains.

**Actual Results**

**Expected results**

**Additional Information**

tonymaglioni, 2019-01-16:
> We've had a number of issues with older drivers, so that might also be the case here. Did you check if they are up to date?

No. I’ll check. Video drivers, correct?

nallath, 2019-01-17:
Yup

tonymaglioni, 2019-01-18:
it said that they are up to date :(