Ultimaker/Cura #5180: [3.6] Print speed calculation error

johnkray, 2019-01-16:
**Application Version**
3.6

**Platform**
Windows 10

**Printer**
Custom printer

**Steps to Reproduce**
Copy all my attached cura config files to the program files for Cura 3.6. Launch cura, add the Hydra Research Nautilus from the add printer menue. Then seletct "normal (.15)" profile for any material.

**Actual Results**
Infill speed, wall speed, and top bottom wall speed do not calculate properly. They do calculate based on print speed but not to the expected values. For all other quality profiles (.1 .2 .3) the speeds do calculate according to the formulas defined.

**Expected results**
The printing speeds should calculate according to the formulas. For my formulas defined in my printer configuration the easiest way to  check if everything is working properly is to set the print speed to a value < 25 which should make all other print speed values that would normally be calculated be = to the print speed.

**Additional Information**
I have tried disableing my custom print speed formulas and using the default cura print speed formulas and for the normal (.15) profile the speed still calculate incorrectly, making me think there is some issue will all the "noirmal (.15)" profiles I have generated. However, I cannot cannot notice any issues with them. 

My custom Cura config files are in a zip at the bottom. I realize that my config is pretty involved, but hopefully some one can help. Thank you in advance!

Expected result with print speed >25 (with "draft (.2)" quality profile selected)
[image]

Expected result with print speed <25 (with "draft (.2)" quality profile selected)
[image]

Unexpected result with print speed >25 (with "normal (.15)" quality profile selected)
[image]

Unexpected result with print speed <25 (with "normal (.15)" quality profile selected)
[image]

All my custom printer configuration files.
[resources.zip](https://github.com/Ultimaker/Cura/files/2762469/resources.zip)

diegopradogesto, 2019-01-23:
As far as I could investigate on this, I see the problem that you are setting the values in your quality profiles (that is correct) but you are still using the default global profiles. In case of the 0.15mm layer height, it's using the `fast.inst.cfg` that defines a different speed for the infill. 

For avoiding confusion, I think it's better for you to define your own global qualities. I tested and that fixes your issue.

BTW, I found that you have a typo in your machine definition file, using `Ture` instead of `True` in two settings ;-)

Hope it helps!

johnkray, 2019-01-23:
That totally fixed it! I was unaware of global qualities. Thanks for taking the time to take a look!

P.S. Thanks for catching my typo ;)

closed 2019-01-24 by Ghostkeeper