Ultimaker/Cura #5182: jerk_travel and acceleration_travel default value being ignored

drzejkopf, 2019-01-16:
jerk_travel and acceleration_travel value cannot be overriden by setting default_value in printer definition, because it is either set to jerk/acceleration_print value when Spiralize outer contour is enabled or to 30 (jerk)/5000 (acceleration) because of "value" parameters of those settings in fdmprinter.def.json:
https://github.com/Ultimaker/Cura/blob/1b7b302f42f82f2a5424d969b370d896febacc33/resources/definitions/fdmprinter.def.json#L3025-L3038
https://github.com/Ultimaker/Cura/blob/1b7b302f42f82f2a5424d969b370d896febacc33/resources/definitions/fdmprinter.def.json#L3305-L3317

I tried to solve this problem, but as I'm not a programmer I haven't suceeded, but maybe someone more experienced will have idea how to make it use print parameter value when magic_spiralize is enabled or default parameter value when it's disabled

smartavionics, 2019-01-16:
I guess you will need to set it in the print profile instead. Here's an example of doing that...

[image]

drzejkopf, 2019-01-16:
@smartavionics I know that, but when developing printer definition I want to have everything set in that definition in order not to make separate set of print profiles for that particular printer. I have removed "value" parameters for both settings in my installation of Cura on my PC, but I gues that if conditional for magic_spiralize exists, ther there is reason for that, so my way of solving that problem is not an ultimate solution

smartavionics, 2019-01-16:
Ah, understood. That conditional was created back in 2016, long before I even knew that Cura existed. I really can't imagine what they were thinking of when they decided to special case accel/jerk when spiralizing. Sorry I can't help.

drzejkopf, 2019-01-16:
@smartavionics Yeah, I know that it was made some time ago, I have tried changing that parameter and it only affected extremely short XY moves, like
```
G1 X105.296 Y113.742 Z82.8 E1265.74963
M204 S2000
;MESH:NONMESH
G0 F3600 X105.295 Y113.741
;TIME_ELAPSED:2157.271740
;LAYER:413
M204 S500
```
so it moved 0.001 mm in X and 0.001 mm in Y (it looks similiar in other parts of gcode), barely any machine is able to move such a small distance, so I don't think it would affect anything at all. Maybe it will affect something on some specific models