Ultimaker/Cura #5186: [master] can't load marketplace right now, is that to be expected?

smartavionics, 2019-01-17:
**Application Version**

**Platform**

**Printer**

**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

I've been getting this recently, is that to be expected at the moment?

[image]

nallath, 2019-01-17:
That's not expected. Weirdly enough, it does work for me on master. 
Is there something in the logs that shows what is going wrong?

smartavionics, 2019-01-17:
Not much, when I click on the marketplace button I get this:
````
2019-01-17 13:45:24,284 - INFO - [MainThread] Toolbox.src.Toolbox._makeRequestByType [553]: Requesting packages metadata from server.
2019-01-17 13:45:24,292 - INFO - [MainThread] Toolbox.src.Toolbox._makeRequestByType [553]: Requesting authors metadata from server.
2019-01-17 13:45:24,294 - INFO - [MainThread] Toolbox.src.Toolbox._updateInstalledModels [275]: Found a plugin that was installed with the old plugin browser: Cura-DuetRRFPlugin
2019-01-17 13:45:24,296 - INFO - [MainThread] Toolbox.src.Toolbox._updateInstalledModels [275]: Found a plugin that was installed with the old plugin browser: FreeCADIntegration
2019-01-17 13:45:24,297 - INFO - [MainThread] Toolbox.src.Toolbox._updateInstalledModels [275]: Found a plugin that was installed with the old plugin browser: MeshTools
2019-01-17 13:45:24,298 - INFO - [MainThread] Toolbox.src.Toolbox._updateInstalledModels [275]: Found a plugin that was installed with the old plugin browser: OpenSCADIntegration
2019-01-17 13:45:24,299 - INFO - [MainThread] Toolbox.src.Toolbox._updateInstalledModels [275]: Found a plugin that was installed with the old plugin browser: PrinterSettingsPlugin
2019-01-17 13:45:24,306 - DEBUG - [MainThread] Toolbox.src.Toolbox._createDialog [225]: Marketplace: Creating dialog [Toolbox.qml].
2019-01-17 13:45:25,034 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [354]: file:///opt/cura/lib/cura/plugins/Toolbox/resources/qml/ToolboxInstalledTile.qml:77:17: Unable to assign QJSValue to QString
2019-01-17 13:45:25,034 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [354]: file:///opt/cura/lib/cura/plugins/Toolbox/resources/qml/ToolboxInstalledTile.qml:77:17: Unable to assign QJSValue to QString
2019-01-17 13:45:25,092 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [354]: file:///opt/cura/lib/qml/QtQuick/Controls/ScrollView.qml:234:9: QML Binding: Binding loop detected for property "value"
````

I have some old plugins still installed but the're all out of date.

stdout/stderr doesn't give anything else.

nallath, 2019-01-17:
As there was only one place where the status was set to errored out without logging, it must be there. I've added the logging just now.
Could you check again with the latest master? It should give a bit more info now.

smartavionics, 2019-01-17:
Done...
````
2019-01-17 15:09:05,309 - WARNING - [MainThread] Toolbox.src.Toolbox._onRequestFinished [652]: Unable to connect with the server, we got a response code 404 while trying to connect to PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v/cura/v6.0.0/packages')
2019-01-17 15:09:05,329 - WARNING - [MainThread] Toolbox.src.Toolbox._onRequestFinished [652]: Unable to connect with the server, we got a response code 404 while trying to connect to PyQt5.QtCore.QUrl('https://api.ultimaker.com/cura-packages/v/cura/v6.0.0/authors')
````

ChrisTerBeke, 2019-01-17:
It's missing the API version number in the URL There's a single 'v' between slashes instead of 'v1'. This value can be set in CuraVersion.py, but should have a sensible ('1') default somewhere.

nallath, 2019-01-17:
Ah, I missed one of the defaults that I changed. The build script sets it to an empty string if it doesn't get data. 72a7cc59cee47301ddbe5cbc861fcad6bd26e589 should fix it.

smartavionics, 2019-01-17:
OK, that nailed it. Thanks guys.

closed 2019-01-17 by smartavionics