Ultimaker/Cura #5188: More settings for multiextruder!

lokanaft, 2019-01-17:
**Application Version**
3.6

**Platform**
**Printer**
multiextruder

**Steps to Reproduce**
use multiextruder

1) **Combing mode**
In any settings it's go over model for another extruder and when you use PVA - it's epic fail. 
Need setting for combing only over this extruder model.
2) **Extrusion cool dow speed modifer**
I Don't know for what it's used more, but for heating and colling extruder should be different settings. Extruder heating more faster, then coolling and now not active extruder forced be heat, for not lost time for it heating.
3) **Extruder change** 
Why it's always change over wall? Why not to change extruder over infill?

rburema, 2019-01-17:
Hi @lokanaft,
Please put each issue/request in a separate github-ticket. Putting multiple items in a single ticket will cause confusion!
cheers, Remco.

smartavionics, 2019-01-18:
Hello @lokanaft , please zip up and attach sample gcode that shows this problem. Also a Cura project file would be good (need to zip or rename .3mf to zip for github to accept it). Thanks.

lokanaft, 2019-01-20:
Image:
[image]
Video:
https://youtu.be/fyPfIEnIZOg

lokanaft, 2019-01-20:
Project, stls, gcode
[Combing.zip](https://github.com/Ultimaker/Cura/files/2776500/Combing.zip)

lokanaft, 2019-01-20:
But without Overlap I will lose strength

smartavionics, 2019-01-20:
Looking at the CuraEngine code, it's not easy to inhibit combing across regions that use different extruders and have been overlapped. Sorry, I don't think the amount of work it would take is worth it.