Ultimaker/Cura #5190: creates incorrect gcode

Aleksandr83, 2019-01-17:
**Application Version**

3.6
**Platform**

windows 10 x64
**Printer**

[form_cream.zip](https://github.com/Ultimaker/Cura/files/2770478/form_cream.zip)

**Steps to Reproduce**

Click the "prepare" button. Click the "Save to File" button. Open the resulting file
**Actual Results**

**Expected results**

**Additional Information**

creates an invalid gcode. Incorrectly formed layers (from 43 to the last).

rburema, 2019-01-17:
Hi @Aleksandr83,
Your model isn't watertight. You can check this in the x-ray view (the red color is where the problems are), but it's visible in normal view as well. Cura doesn't support non-watertight models.
cheers, Remco.

Aleksandr83, 2019-01-22:
I understood everything, thanks for the answer!

closed 2019-01-22 by Aleksandr83