Ultimaker/Cura #5192: [3.6.0] High CPU usage on idle

crazzzik, 2019-01-17:
**Application Version**
[3.6.0]

**Platform**
Windows 10

**Printer**
Prusa i3 over Octopi

**Steps to Reproduce**
-Start application
-Add model
-Send Print
-Wait until print finished
-Observer CPU usage

**Actual Results**
Cura uses between between 14% and 40% CPU doing nothing on Prepare, Monitor or even background.

**Expected results**
Expected to go to idle CPU consumption (0-2%)

**Additional Information**

[image]

smartavionics, 2019-01-18:
There was a recent issue where the engine would sit using 100% CPU after cura started. Once the first slice had been done, the CPU usage dropped to 0 (in the engine). Maybe it is that you are seeing. Currently, on a Linux cura, the idle CPU usage is low unless I waggle the mouse around over the cura window when it uses more.

nallath, 2020-11-26:
This issue has been fixed in the meantime.

closed 2020-11-26 by nallath

Venryx, 2021-04-26:
In Cura 4.9.0, I'm still noticing a 5-6% continuous CPU usage, even when Cura is without a loaded model + defocused + minimized.

Not as bad as 30%, but a 5% continuous load when it should be ~0% is still worth fixing. (and my processor is 8-core, 16-thread, so it's actually using half a core for doing nothing [or at least it should be])