Ultimaker/Cura #5198: delete key doesn't work

jhihn, 2019-01-19:
**Application Version**
3.6.0

**Platform**
OSX (High Sierra and Mojave)

**Steps to Reproduce**
Add an item to the build plate. Then try to remove it with the delete key. It does not delete. 

**Actual Results**
Nothing happens

**Expected results**
The selected item is removed.

**Additional Information**

LipuFei, 2019-02-21:
Closing this due to inactivity.

jhihn, 2019-02-21:
Sorry. I confirmed this. Still an issue for me.

Ghostkeeper, 2019-02-22:
We're still unable to reproduce the issue though.

jhihn, 2019-03-01:
It worked like once, I was surprised, then stopped working for me.  I'll keep playing, but it is a thing.
Is there any debugging I can enable?

Ghostkeeper, 2019-03-04:
Not really. We log the highest logging level anyway in Cura's logs. Maybe it helps to share the Cura log? I'd be surprised if it mentions something though.

jhihn, 2019-03-16:
Video:
https://drive.google.com/open?id=1q9LxxsmhXTOb3vbyHoL5l_-UM9zXOcM2

Ghostkeeper, 2019-03-19:
I believe you, don't worry. But if we can't reproduce the problem on this side we're just stabbing in the blind and can't really fix anything for you.

jhihn, 2020-06-20:
Still a thing in 4.6.1

Ghostkeeper, 2020-06-30:
Yup. There has been some more investigation in https://github.com/Ultimaker/Cura/issues/7754 though, so let's consolidate the information there. I'll close this as a duplicate.

closed 2020-06-30 by Ghostkeeper