Ultimaker/Cura #5209: Cura 4.0.0 beta - Z Seam Relative not correctly working

amagr0, 2019-01-22:
**Application Version** - CURA 4.0.0 beta
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

**Steps to Reproduce** - Open Cura Project; Slice; analyse the toolpath.

**Actual Results** - Different Z seam alignment on parts.

[image]

[z_seam_aligm.zip](https://github.com/Ultimaker/Cura/files/2782526/z_seam_aligm.zip)

smartavionics, 2019-01-22:
Hello @amagr0 , please provide the project file that includes the settings and not just the model. Thanks.

smartavionics, 2019-01-22:
Ah, I think I know what is happening. Those 6 items have been grouped into a single mesh so the z-seam is relative to the whole bounding box and not the bounding boxes of the individual parts. You need to ungroup them.

amagr0, 2019-01-22:
Hi @smartavionics,
By mistake I exported the model, and not the project.
I normally use Netfabb to organize multiple parts that could be till 100 by bed, Cura is not so effective to deal with so many parts, and then I export a single stl file to Cura with all the parts.

I will start to export in 3mf format, this seems to solve this problem.
[image]

[Proj_z_seam_aligm_3MF.zip](https://github.com/Ultimaker/Cura/files/2783122/Proj_z_seam_aligm_3MF.zip)

Thanks

smartavionics, 2019-01-22:
That looks better, please close this issue if you are now happy.

closed 2019-01-22 by amagr0