Ultimaker/Cura #5210: Cura 3.6.0 - Add new natively printer

amagr0, 2019-01-22:
**Application Version** - CURA 3.6.0
**Platform** - Win 10 Pro, 64 bit
**Display Driver** - NVIDIA GeForce 840M

How can I change this perimetral area?
[image]

I already looked for some variables in fdmprinter.def file and change it on new definition printer, but it seems to not take any effect.

Thanks

nallath, 2019-01-23:
Those areas are generated because of the platform adhesion. As platform adhesion is not per object (eg; it doesn't prevent objects from getting closer to eachother), we apply it at the edge of the build plate (as that space is needed for the platform adhesion, it can't contain an object). 

If you want to add aditional no print areas, you need to use "machine_disallowed_areas", where you can set a list of polygons that need to be added.

closed 2019-01-23 by nallath

amagr0, 2019-01-23:
Thanks @nallath