Ultimaker/Cura #5214: Cura 3.6.0 unable to send print to Octoprint

kmalherbe, 2019-01-23:
Hello,
I am brand new to this site, and am really new to 3D printing, but I have managed to turn out some really nice stuff and configure my printer quite well, however...

I am experiencing an issue with sending prints to Octoprint from Cura 3.6.0. Everything else is communicating fine. I can control my printer from BOTH Cura and Octoprint via Pi 3B+. This includes all steppers, extruder, heatbed, etc. Everything worked just fine last night, but I can't get my prints to send today. After slicing, preparing, and clicking on "Print with Octoprint" tab, a message appears on the bottom of the Cura window reading "OctoPrint returned an error: CONFLICT." Again, everything else is working properly without issue.

The only change I have made, was an update on the Octoprint side this morning. Since then, no prints have been able to transfer.

I have rebooted and reconnected several times. I have searched for firmware updates in Cura 3.6.0. I have restarted the Pi, printer, and wifi multiple times as well. Still everything reconnects without issue, but I am still unable to send prints from 3.6.0 to Octoprint.

Any suggestions are greatly appreciated! And thank you in advance for your time!

**Application Version**
Cura 3.6.0

**Platform**
Windows 7

**Printer**
Anet A6

kmalherbe, 2019-01-24:
Update:

I decided that I would attempt a full uninstall of the Cura 3.6.0 software. After reinstallation, configuration, and Octoprint Comm plug in install, everything appears to be working normally again. Not sure what the issue was, but it was indeed frustrating. 

Hope this helps someone else down the road!

Thisismydigitalself, 2019-01-24:
I had the same issue two weeks ago. uninstalling and reinstalling only octoprint 3.5.3 plugin was enough to get things going again.

kmalherbe, 2019-01-24:
Good to know. When I did the reinstall, I used version 3.5.1, but then
installed the 3.6.0 when prompted in the software. I guess you could say
the "issue is resolved"? I'm just glad it's working!

fieldOfView, 2019-01-24:
I doubt reinstalling the OctoPrint Connection plugin actually did the job. 

That unhelpful "CONFLICT" message can mean a number of things, all saying that OctoPrint was unable to store the uploaded gcode file. These are a couple of possible reasons:
* You are trying to upload a gcode file with the same jobname as the file that is currently being printed
* You are trying to upload to the printer sd card, but there is no sd card in the printer or the card is full
* Storage on the octoprint drive is full

The plugin showing the "CONFLICT" message is actually a bug; it should show the actual human readable part of the error message returned by OctoPrint. This should be fixed in the next version of the OctoPrint Connection plugin which will be available in a week or so.

For future reference, please report issues with the OctoPrint Connection plugin here:
https://github.com/fieldofview/cura-octoprintplugin/issues
The OctoPrint Connection plugin is a 3rd party plugin, and the Ultimaker team won't be able to help you much with it. Also, I cannot close the issue here once it is resolved.