Ultimaker/Cura #5217: [Cura 4.0] Nozzle Option Randomly Missing

Enverex, 2019-01-24:
The nozzle option tab under the machine page of the settings menu only seems to appear under certain circumstances. The most reliable way to get it to appear is to add a second dummy printer, which for some reason allows the tab to appear. Here's the settings menu normally, note the missing tab for the nozzle:

[image]

Now, after adding a second printer (but keeping the original one that we want to use as active), the tab appears:

[image]

I've also seen the nozzle option appear without having to add a second printer, but that was rare and I'm not sure what triggered it to appear in those scenarios.

paukstelis, 2019-02-01:
Confirmed for me as well. Trying to change imported printer from 2.85 to 1.75 and simply can't do it via the gui.

Ghostkeeper, 2019-02-06:
I also can't reproduce it with the Wanhao Duplicator 6. My reproduce steps:

* Clear settings completely.
* Start Cura and accept the license agreement.
* Select Wanhao Duplicator 6 and click add.
* Go to the machine manager and click the Machine Settings button.

The extruder tab appears fine, even though the Duplicator 6 is the only printer. Also after restarting Cura, the extruder tab is present. Maybe something was fixed since the first beta release? Anyone knows?

Enverex, 2019-02-06:
I'm running a GIT master build on another machine and I'm not seeing the problem there. The nozzle tab appears after about 0.2 seconds (I assume that's the plugin loading?) so it looks like it's been fixed at some point. I'll let others confirm when it trickles down to actual releases.

nallath, 2019-02-06:
I vaguely recall that we did some fixes with regards to that code, but I don't remember well enough to be able when and for what issue.

GregValiant, 2026-09-07:
I'm cleaning up these old reports and requests.  I'll go ahead and close this.  If the problem still persists in 5.13.0 then please make out a new report so it goes to the top of the pile again.

closed 2026-09-07 by GregValiant