Ultimaker/Cura #5218: [Cura 4] Fix for one USB printing bug

mamaich, 2019-01-24:
Cura 3.6 and 4 is unable to print on some devices (like 101hero). 
The fix is simple - you've forgot to zero _firmware_idle_count in constructor of USBPrinterOutputDevice (file plugins\USBPrinting\USBPrinterOutputDevice.py). 

The second problem is that there is no exception handling inside while loop of _update function of the same class. So the function terminates if an exception is thrown in one of the called functions, causing interrupted prints. 

Fixed file is attached (remove the .txt extension), or you can apply the DIFF:
<pre>77a78,79
>         self._firmware_idle_count = 0
>         self._firmware_name = None
225a228
>           try:
229c232,233
<                 continue
---
>                 line = b""
> #                continue
309a314,316
>           except:
>             Logger.log('e', "Exception in _update()")
>             continue
</pre>
[USBPrinterOutputDevice.py.txt](https://github.com/Ultimaker/Cura/files/2791687/USBPrinterOutputDevice.py.txt)

Ghostkeeper, 2019-01-25:
Indeed the `_firmware_idle_count` should always get initialised in the `__init__`. It's dangerous not to. I fixed that for you. Thanks for investigating!

I didn't put the try-except there because if the connection is busy for some reason we don't want to pump it full of g-code commands. It may be something we could take a look at later at some point, because you could catch some exceptions a bit more specifically.

closed 2019-01-25 by Ghostkeeper

Palmfan, 2019-02-01:
Thanks a lot mamaich, you're a genious, I was investigating for some days that same problem with Cura 4.0 and Alfawise U20 and…  you are the best ! I just dowloaded your file and IT WORKS !
Best regards.
Palmfan

commit 91dc3713 (2019-01-25): Initialise idle count properly

This is a field so it should always get initialised in the __init__ function.

Fixes #5218.