Ultimaker/Cura #5219: Loading gcode doesn't view initial layer

BagelOrb, 2019-01-24:
**Application Version**
Cura 4.0.0 BETA

**Platform**
Linux MInt

**Printer**
UM3

**Steps to Reproduce**
Load attached gcode
[Voronoi_2.gcode.zip](https://github.com/Ultimaker/Cura/files/2792186/Voronoi_2.gcode.zip)

**Actual Results**
[image]

**Expected results**
[image]
The brim should be shown

**Additional Information**
Generated using surface mode if I remember correctly, but I can't imagine that's relevant.
Generated using older version of Cura for an UM3.

Ghostkeeper, 2019-01-25:
It doesn't seem to happen with your g-code in the current state of 4.0:
[image]

I can't reproduce this. Maybe it has been fixed in the meanwhile?

BagelOrb, 2019-01-25:
I'm so damn stupid.

I most probably just had Helper parts disabled in the layer view.

I keep getting confused by the layer view. 
Because when you scroll through a single layer everything is visualized I keep forgetting that I have disabled the visualization of helper parts.
Even when you have disabled the visualization of e.g. helper parts they will still shown when scrolling through a layer with the horizontal play bar.
Would you consider that as a bug?

Ghostkeeper, 2019-01-25:
> Would you consider that as a bug?

Sorta yes. I knew about it, but wasn't bothered enough to really do something about it.

BagelOrb, 2019-01-25:
Yeah same here.