Ultimaker/Cura #5222: [4.0 Beta] Tag beta releases on Github

tetov, 2019-01-25:
**Application Version**
4.0

**Platform**
Arch Linux 

**Additional Information**

I maintain PKGBUILDs in the [AUR](https://aur.archlinux.org/) for the beta branches of [cura](https://aur.archlinux.org/packages/cura-beta-git/), [curaengine](https://aur.archlinux.org/packages/curaengine-beta-git/), [libarcus](https://aur.archlinux.org/packages/arcus-beta-git/) and [uranium](https://aur.archlinux.org/packages/uranium-beta-git/) for installation on Arch Linux. They build from the latest commit on the 4.0 branches, which might be interested for some users who want to use the absolute latest version.

However, it would be nice to be able to package the beta release instead, the version you have released as an AppImage.

Would it be possible for you to tag the beta releases the 4 projects on GitHub so we can use that as a source instead?

Ghostkeeper, 2019-02-14:
Possible, yeah, but so far we haven't even tried to. We quite often forget to tag even the actual releases! Sorry :( I'll see what I can do.

tetov, 2019-02-16:
Thank you! I’ll keep an eye out :).

Ghostkeeper, 2019-02-26:
I've tagged Cura, cura-binary-data, cura-build, cura-build-environment, CuraEngine, fdm_materials, libArcus, libCharon, libSavitar and Uranium (all the repositories that you'd need to build Cura completely except the firmware source) with a 4.0-beta tag. This tag refers to the second beta that we released today.

I didn't tag the Marlin repositories that we ship with Cura. They haven't changed in a while and also aren't re-built for this Cura release. The version that we ship is included in cura-build though.

closed 2019-02-26 by Ghostkeeper