Ultimaker/Cura #5223: [3.6] Supports cross the model and go inside

Liger0, 2019-01-25:
**Application Version**
3.6

**Platform**
Win10 x64

**Printer**
Custom dual extrusion

**Steps to Reproduce**
Load the project

**Actual Results**
The infill goes into the model

**Expected results**
The infill should be separated from the model

**Additional Information**
[image]
[supports in a model.zip](https://github.com/Ultimaker/Cura/files/2798232/supports.in.a.model.zip)

smartavionics, 2019-01-26:
I could not import your settings from the project because the printer was unknown to my cura. I did get the model which sliced OK using my settings. Could you please provide another project that has this problem and uses a known printer type. Thanks.

[image]

Liger0, 2019-01-26:
> 
> 
> I could not import your settings from the project because the printer was unknown to my cura. I did get the model which sliced OK using my settings. Could you please provide another project that has this problem and uses a known printer type. Thanks.
> 
> [image]

Could you try with a double extrusion 1.75 filament? nozzle distance is 30mm.
Mine is a modded cr-10s with double extrusion definition.

smartavionics, 2019-01-27:
Even with tree supports enabled, I do not get this problem. Does it only happen with tree supports?

Liger0, 2019-01-27:
> 
> 
> Even with tree supports enabled, I do not get this problem. Does it only happen with tree supports?

Yes.

Ghostkeeper, 2019-02-14:
I'm also unable to reproduce this on a different multi-extrusion printer (UM3). I tried setting the support extruder to something else than the build extruder, and setting both extruders to the same extruder. I couldn't get the support to clip the model.

And I tried setting the filament diameter to 1.75 in Custom FFF Printer but that obviously didn't make any difference to layer view.

Liger0, 2019-04-03:
It still happens on cura 4.0.
I have saved the project file but I can't share it publically because it's not a public stl, so let me know how can I share it to you. It happens along with a strange point where the tree supports starts in the air.
This time I just used a basic custom dual extrusion machine done directly by Cura itself and a new profile, so everything is fresh, and the bug persists.
[image]
[image]

Ghostkeeper, 2019-04-03:
You could send a project file to me on r.dulek at ultimaker.com. I swear to share it only with colleagues working on Cura, and only for debugging purposes!

Liger0, 2019-04-03:
> 
> 
> You could send a project file to me on r.dulek at ultimaker.com. I swear to share it only with colleagues working on Cura, and only for debugging purposes!

I tried to sent you a message: "r.dulek cannot receive messages."

Ghostkeeper, 2019-04-04:
Oh, eeh, that is a serious concern if I can't receive messages from outside then. I'll have IT look into this. You can try to send it to my personal one at rubend at tutanota.com then.

Liger0, 2019-04-04:
> 
> 
> Oh, eeh, that is a serious concern if I can't receive messages from outside then. I'll have IT look into this.
> 
> EDIT: I can send e-mails from my personal account to my work account. I think you may have typed some e-mail address wrong then?

I just used the function to send a message to a user using the nickname.

Ghostkeeper, 2019-04-04:
You can send PMs to people in Github? I didn't know that and I can't find it. Please use e-mail instead.

Liger0, 2019-04-04:
> 
> 
> You can send PMs to people in Github? I didn't know that and I can't find it. Please use e-mail instead.

I was referring to Ultimaker forum. How can I send you an e-mail?

Ghostkeeper, 2019-04-05:
Ah, I understand the mistake. r.dulek at ultimaker.com was meant to be my e-mail, not my Ultimaker forum username. My username on the Ultimaker forum is also Ghostkeeper. Either one works.

Liger0, 2019-04-05:
I didn't know a name without the @something.something would be an email. By the way, I've sent the message.

Ghostkeeper, 2019-04-11:
The issue is reproduced with this project file. Strange how the support does have a proper border on the inside but the infill lines don't stop there.

Liger0, 2019-04-11:
The dot on the bottom in the air is also something I noticed very frequently with cura 4.0

Ellecross, 2019-04-15:
Devs see CURA-6465.

Ghostkeeper, 2019-06-12:
Nothing changed recently in this ticket, no. Sorry. It's still on the ToDo list.

Lieste, 2020-02-11:
I had this recently with a model, it happened at the layer where the tree supports completely wrapped the model, and lasted for the height where the ring was complete. I had a properly formed ring of infilled support, as well as an additional disk with infill.
If I slightly reduced the tree support branch angle, the ring opened and the support didn't cross into the model, but I also lost supports over around half the target overhangs... https://www.myminifactory.com/object/3d-print-spring-frost-101774 set to 355mm high, supports touching base, I only get supports reaching the buttock overhang at support branch angle =55 not at 50. At 55, I get the failure at layer ~1210 (0.2mm) where the ring of tree supports forms a complete circle, and the infill covers the outer ring completely, instead of only the portion between inner and outer walls.

nallath, 2020-07-01:
This has been fixed for 4.7

closed 2020-07-01 by nallath