Ultimaker/Cura #5226: Z-hop not working with combing mode over max distance

bealparslan, 2019-01-26:
Settings:
- Combing mode: on
- Max comb distance with no retract: 5mm
- Z Hop: on
- Z Hop height: 0.6mm

In a printing; if travel distance over 5mm, retracting happened but **z-hop not working.**

RonnieJHW, 2019-12-05:
Why was this closed.  I have the same problem on many prints.  My only remedy (I found so far) is to turn off combing, then I do get a z hop when retracting.
Using Windows 7, Cura 4.4.0 (several flavors in the 4.x.x exhibit the same behavior) and delta style printer (SeeMeCNC Artemis 300).
Last noticed this on a print created with Cura 4.4.0, Z-hop on retract is on, max comb distance is 5 mm, hop height is 2.0 mm, infill is cubic at 20%.
I can see the printer retract, but no hop at all and hits all the infill lines.

Ghostkeeper, 2019-12-06:
Can you try disabling "Only Hop Over Printed Parts"?

RonnieJHW, 2019-12-06:
That is already disabled.  Thanks for looking at this ticket.

Full listing of Travel settings:

Combing: All
Max Dist w/ no retract: 5 mm
Retract before wall: off
Avoid Printed parts when traveling: on
Avoid supports when traveling: off
Travel avoid distance: 3 mm
Layer start X: 0.0
Layer start Y: 0.0
Z hop when retracted: on
Z hop only over printed parts: off
Z hop height: 2 mm
Z hop after extruder switch height: 2 mm

Ghostkeeper, 2019-12-09:
Looking more closely at your bug report and your settings, I think you mean to disable combing. Combing instructs the printer to avoid hitting the walls, and the idea is that then it will not be necessary any more to retract (or Z Hop). However since your Max Distance with no Retraction is set to 5mm only, you will still get a retraction, just without a Z Hop.

Here is a bit of documentation on travel moves and when it should be retracting, hopping or combing: https://github.com/Ultimaker/CuraEngine/blob/CURA-6860_fix_combing_retraction/docs/assets/retraction_combing_hop.svg

RonnieJHW, 2019-12-09:
Thanks much for the documentation, now things make sense.

I would still have liked/was expecting for the Yes branch of the "longer than max distance" to feed into the "Z hop enabled" question.  What I was hoping is to have an outcome where it would always perform a Z-hop when retracted, even with  combing (as the comment on z-hop suggests).

Actually, during the infill, I would love to have an option to do a z-hop without a retract (combing infill enabled).  This would have the benefit of the nozzle not rubbing during travel moves across the infill, but not do a retract so less risk of grinding down the filament.  I want to leave combing on to limit the amount of wall crossings and prevent stringing, especially when printing PETG...

Ghostkeeper, 2019-12-10:
> What I was hoping is to have an outcome where it would always perform a Z-hop when retracted, even with combing (as the comment on z-hop suggests).

In that case you'd move up above the walls you need to avoid and then still avoid them. So you might as well go straight then. There is no reason to avoid walls horizontally if you're already avoiding them vertically. So combing + z hop is not a possible outcome.

RonnieJHW, 2019-12-10:
I respectfully disagree for two reasons: 1) stringing when combing is off where the print head moves straight across holes, I find material in open spaces all the time (I typically print engineering pieces with a significant amount of open features) and 2) print head picking up gunk (especially when printing PETG) when running across 50 mm or so of infill at travel speed, that usually gets deposited somewhere unfortunate.

But I get your point that not everybody wants this feature and the reasoning behind not having this as a generally useful behavior.  I'll look into making a plug-in.  Thanks so much for indulging my curiosity!

Ghostkeeper, 2019-12-11:
With 1, combing will not solve the issue since the stringing gets pulled over the hole afterwards, since it doesn't hit anything until the nozzle lands.

2 should not happen because your print head is moving over the infill rather than through it. If it's hitting the infill, the Z hop height needs to be increased.