Ultimaker/Cura #5228: infill mesh issue

pektrus, 2019-01-28:
i have a problema with my settings when i do mesh type
when i selecte  "modify settings for infill other models, i can´t find "infill mesh" and is so anoying … i dont know to do

i already tried go to the settings and apears but when i do the "modify settings for infill … " doesn´t apears  again… just apears "infill mesh order" … i dont know what to do

please help me!!

pektrus, 2019-01-28:
i have the 3.6 upadate (november 2018)

fieldOfView, 2019-01-28:
If you select "Modify settings for infill of other models" as mesh type, that adds the infill mesh setting for you. You don't need to (and can't) set it any more than that.

fieldOfView, 2019-01-29:
Attachments don't come through when posting to github. Please add them via https://github.com/Ultimaker/Cura/issues/5228

Ghostkeeper, 2019-01-29:
If you "Save" your model it should retain stuff like the infill meshes. If you "Export" your model it saves the file in a format that is used for exchanging with other 3D model applications. It will lose your settings then (including which models are infill models).

pektrus, 2019-01-29:
yeah but i don´t what happen
i save and when i open again, the model that i used to put diferentes
infill meshs , joint and became one piece …
so i dont know what i should do
please help

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Sem
vírus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

à(s) 15:15:

Ghostkeeper, 2019-01-29:
Maybe you can upload that file on github.com in your issue?

closed 2019-01-29 by pektrus

pektrus, 2019-01-29:
guys
i already out my problema in the site
https://github.com/pektrus/Cura/tree/pektrus-infill-mesh-save-issue

thanks

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Sem
vírus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

à(s) 15:34:

pektrus, 2019-01-29:
the tail just put 1000% (dimesnion) in z,x,and y

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Sem
vírus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

SimãoManuel SUP <s.manuel@campus.fct.unl.pt> escreveu no dia terça,
29/01/2019 à(s) 16:05:

pektrus, 2019-01-29:
my problema is why the object joint to the model that i used to do
diferents infill mesh
i got some pictures
this pictures are the before and the after (after i save and open again)

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Sem
vírus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

SimãoManuel SUP <s.manuel@campus.fct.unl.pt> escreveu no dia terça,
29/01/2019 à(s) 16:15:

pektrus, 2019-01-29:
my problema is why the object joint to the model that i used to do
diferents infill mesh
i got some pictures
this pictures are the before and the after (after i save and open again)
[image]
[image]

Ghostkeeper, 2019-01-30:
The files you uploaded are only STL files. These do NOT contain any Cura-specific information such as whether the object is an infill mesh or not. If you want to retain this information, you need to save it as a Cura project file, which is under `File` -> `Save...`, (NOT `File` -> `Export...`).

This seems to be behaving as expected, so far.

pektrus, 2019-01-30:
thanks so much
it really worked
now, i wuant to pass this file in * 3 mf to solidworks
its easy right? or I have to pass for STL file?

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Sem
vírus. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

à(s) 08:53:

Ghostkeeper, 2019-01-31:
SolidWorks will not be able to interpret nor generate the "infill mesh" setting from Cura. It has no such notion. SolidWorks will read the 3D data from the 3MF file, but will not mark any mesh as being an infill mesh or anything.

pektrus, 2019-01-31:
There is any program that can read the infill mesh and simulate forces ??

:

Ghostkeeper, 2019-01-31:
Not as far as I know.

When you save a project from Cura (using `Save...`) it saves a 3MF file, which is actually a zip file (try opening it with an archiver application) but apart from the normal 3D data we also save the profiles and per-object settings. This is saved in Cura's own configuration format. As far as I know, no other application was ever implemented to read this format.

Ghostkeeper, 2019-01-31:
You would need to ask SolidWorks to implement that though. SolidWorks doesn't implement Cura's infill patterns, infill layer thickness, infill printing speeds, etc. so it won't be able to simulate forces on it.

You could try converting the g-code to a 3D model and simulating forces on that with FEM, but it'll be very slow and inaccurate.