Ultimaker/Cura #5230: Zeroconf 0.17.6 cannot handle service names with periods

fieldOfView, 2019-01-28:
**Application Version**
Up to 4.0

**Platform**
any

**Printer**
any, OctoPrint

**Steps to Reproduce**
In OctoPrint, preferences, appearance, set the instance name to "Test.test". Restart the OctoPrint instance

**Actual Results**
The instance will no longer be discoverable because the zeroconf ServiceBrowser chokes on the . in the servicename with a zeroconf.BadTypeInNameException exception.

**Additional Information**
I am working around this issue by including an updated zeroconf.py version with my plugin (0.21.3 instead of the version 0.17.6 that is included with official builds). This fixes the problem for OctoPrintPlugin, but it may become an issue for other plugins (or even UM3Networking) if there are ever dots in the UM3/UMS5/UMnext printer network names.

fieldOfView, 2019-01-28:
Also see the upstream issue here: https://github.com/jstasiak/python-zeroconf/issues/74

LipuFei, 2019-01-28:
@fieldOfView A tricky problem is that `zeroconf > 0.17.6` seems to be extremely responsive to mDNS network activities. In a network with a lot of mDNS activities, the Cura python process can be constantly on high CPU usage, which makes Cura impossible to use. See this upstream issue for some details: https://github.com/jstasiak/python-zeroconf/issues/133

fieldOfView, 2019-01-28:
Ah, we're between a rock and a hard place then.

LipuFei, 2019-01-28:
I was looking into https://github.com/hashicorp/mdns At some point, I would like to try building a separate process to replace zeroconf.

fieldOfView, 2019-01-28:
> I would like to try building a separate process to replace zeroconf.

That sounds a bit like throwing out the baby with the bathwater. I should be having access to a university network with - I would guess - enough mDNS traffic to see if I can hunt down this issue.

fieldOfView, 2019-01-28:
For my reference, https://github.com/jstasiak/python-zeroconf/commit/e46af83d35b4430d4577481b371d569797427858 fixes the issue with dots.

tritone11, 2019-01-29:
I am getting the same error with this ServiceInfo
```
info = ServiceInfo("worker._tcp.local.",
                       "worker._tcp.local.",
                       socket.inet_aton("127.0.0.1"), 8010, 0, 0)

```
everything looks fine to me but still 
```
File "/usr/local/lib/python3.6/dist-packages/zeroconf.py", line 1407, in __init__
    raise BadTypeInNameException
zeroconf.BadTypeInNameException

```

fieldOfView, 2019-01-29:
@tritone11 note that this issue queue is for Ultimaker Cura, which includes python-zeroconf 0.17.6. Unless you are running into this error when running Cura, I suggest you update your zeroconf install with ```pip install zeroconf==0.21.3```

tritone11, 2019-01-29:
@fieldOfView I am running that exact version unfortunatly, but it doesn't accept even a simple service name like this 
```
info = ServiceInfo("T._tcp.local.",
                       "_T._tcp.local.",
                       socket.inet_aton("0.0.0.0"), 8010, 0, 0)

```

fieldOfView, 2019-01-29:
Then you are not using it in Cura (because Cura uses a different version), so I suggest you visit the python-zeroconf issue queue instead:
https://github.com/jstasiak/python-zeroconf/issues

tritone11, 2019-01-29:
@fieldOfView Very sorry, wrong place to place my issue

LipuFei, 2019-01-29:
> 
> 
> Was `zeroconf==0.17.7` tested? It fixes the `.` issue.

@fieldOfView I'm not sure, but I'll test with 0.17.7 in case it works. Thanks!

LipuFei, 2019-01-30:
For dev, JIRA ticket: CURA-6163

@fieldOfView FYI, we decided to update zeroconf to 0.17.7 for master, but not for 4.0.

fieldOfView, 2019-01-30:
I am including a copy of zeroconf 0.17.7 with the new version of OctoPrintPlugin, so the fix should work for the OctoPrint connection in Cura 3.5 - Cura 4.0.

closed 2019-01-30 by fieldOfView

fieldOfView, 2019-04-09:
Aww, thanks for testing. 
I'll see if I can cherry pick the changes I need to handle service names with periods in the version of zeroconf that I ship with the OctoPrint plugin.