Ultimaker/Cura #5232: Cura 3.6 Relative Extrusion with Smoothieware

Thisismydigitalself, 2019-01-28:
**Application Version**
3.6

**Platform**
Win10/64

**Printer**
Custom printer powered by Smoothieware running on MKS Sbase 1.4, E-Stepper 1.8 deg @ 500mA.

**Steps to Reproduce**
Check "Relatice Extrusion"

**Actual Results**
Seems like Retraction speed is much hight than when not checked. 
My retraction speed is set to 20mms which is much lower than the maximum my E stepper can handle but when i set Relative Extrusion motor stalls - very short squeak sound is heard. Print works great and no print artifacts is visable but system works not as expected. i can't measure actual speed but it feels like stepper is spinning twice as fast., again - not sure. nothing scientific about my findings as i can't measure actual motor speed.

having said that and without changing retraction speed:

When "Relative Extrusion" is NOT checked - all's good.
When "Relative Extrusion" is Checked - Print is OK but Motor briefly stalls and looks like spinning much faster.

Not sure if this is Cura issue or Firmware issue.

**Expected results**
E-Stepper should not stall (only guessing ..) whether Relative Extrusion is checked or unchecked.

Ghostkeeper, 2019-01-29:
Maybe it's forgetting the F in the entry when doing relative retracts?

Thisismydigitalself, 2019-02-02:
Thanks guys. I was away from my desk so only now I'm back for more tests.
I bumped up retraction speed to 30mms (Was 20mms) and lowered E-Acceleration by 50% and no more motor stall in both relative & absolute modes . i have no idea how come this happened cause these settings worked in the past. it might be due to overall speeds settings which reached a threshold.

smartavionics, 2019-02-02:
OK, if you're happy, please close this issue.

Thisismydigitalself, 2019-02-02:
Either i can't or dono how to close my issue(s).

smartavionics, 2019-02-02:
Isn't there a button below next to the comment button that says something like (comment and close)?

smartavionics, 2019-02-02:
Actually, looks like this:

[image]

CirnoGiovanna, 2021-03-30:
Sorry to comment here, but I met problems related to this issue.
For some reason, I want generate gcode with relative extrusion (M83) by Cura. However, I do not found the corresponding useful options in Cura, or in def.json of CuraEngine. After I set the `Relative Extrusion` option visible in Cura, this option is still invisible in the Print setting palette.

[image]

According to this issue, it seems like there are relative/absolute extrusion options in Cura 3.6. Is this option no more supportable in Cura 3.7.1?

Ghostkeeper, 2021-04-15:
If you happen to be using the Anet 3D printer profile, it seems that definition explicitly hides the setting so you can't use it. I suppose the printer doesn't support it. Other than that, the setting should always be made available if it's visible in that menu. For me it works in any case, so I can't reproduce your bug.

Cura 3.7.1 doesn't exist. After Cura 3.6 we released Cura 4.0.