Ultimaker/Cura #5242: [v3.6.0] Post scripts are not executing at all

SourceCipher, 2019-01-29:
**Application Version**

v3.6.0

**Platform**

Win 10 x64

**Printer**

Tevo Michelangelo

**Steps to Reproduce**

Add a script like "Filament Change" or "Pause at Height" from the preloaded scripts. Set to stop or change at layer height, close the popup window, slice your current model and start printing via usb.

**Actual Results**

Nothing happens, no matter what layer I specify its never pausing at the layer or any layer in this case

**Expected results**

To stop printing at the specified layer

**Additional Information**

Error logs for the last print with change filament scrip:

`2019-01-27 23:11:18,823 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/SidebarHeader.qml:551:9: QML QQuickItem: Binding loop detected for property "width"`

`2019-01-27 23:11:18,915 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Menus/ConfigurationMenu/PrintCoreConfiguration.qml:73: TypeError: Cannot read property 'name' of null`

`2019-01-27 23:11:30,830 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/resources/qml/Menus/ConfigurationMenu/PrintCoreConfiguration.qml:73: TypeError: Cannot read property 'name' of null`

`2019-01-27 23:11:30,877 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [352]: file:///C:/Program Files/Ultimaker Cura 3.6/qml/QtQuick/Controls/Styles/Base/ScrollViewStyle.qml:384: ReferenceError: __styleData is not defined`

Vandrasc, 2019-01-30:
Hi @SourceCipher, both plugins work as intended for me, can you please share the project file with us so we can see what went wrong? Thanks!

SourceCipher, 2019-01-30:
@Vandrasc I made the circle with Fusion 360 to test ironing etc. This is the .stl file I tried the scripts with. Link to the file -> https://we.tl/t-RY6WVopZry (weTransfer)

Vandrasc, 2019-01-30:
I do not think it is the model (thank you for sharing though), I am afraid it may be your configuration. We do not have a Tevo Michelangelo in Cura, so, unfortunately, I cannot test. I did try with the other Tevo's supported by Cura and the post-processing plugins work for me.  Would it be possible for you to try with a different printer and see if that works?

SourceCipher, 2019-01-30:
All the settings works perfect with this printer. It prints very well. I dont have any other ones to test on. I will try maybe another model today but I doubt that will change.

Vandrasc, 2019-01-30:
What I meant is if you could add a different printer in Cura and slice with one of the plugins and see if the Pause at Height or Filament change is mentioned in the gcode output. I am sure your printer works just fine. 

@Ghostkeeper , do you maybe have another idea?

SourceCipher, 2019-01-30:
@Vandrasc Oh I see what you mean. I will try that this evening. Will let you know the outcome

SourceCipher, 2019-01-30:
@Vandrasc Just tried with Tevo Tornado which seems to be supported printer. Having the same issue.

closed 2019-01-31 by nallath

Ghostkeeper, 2019-02-01:
Nallath found yesterday that USB printing did not execute post-processing scripts. It would work if you used an SD card or something to put the print in the printer, but not via the USB cable.