Ultimaker/Cura #5245: Missing infill with certain Part

PhilmacFLy, 2019-01-30:
**Application Version**
3.6.0

**Platform**
Windows 10 and MacOS 10.14.2

**Printer**
Prusa i3 Mk2

**Steps to Reproduce**
Try to prepare the file with recommended settings and any infill

**Actual Results**
Infill is missing

**Expected results**
Infill should be there

**Additional Information**
If I print one Rupee and not 12 Rupees on the Plate the infill is generated
[Models.zip](https://github.com/Ultimaker/Cura/files/2812080/Models.zip)

Vandrasc, 2019-01-30:
Hi @PhilmacFLy , this is weird, I can see the hallow model and it seems it also has two outer walls... but I cannot reproduce from scratch when multiplying the single model. How did you create the hollow mesh (with 12 rupee)? I cannot ungroup or select any of them. 

Apparently, there is something wrong with the hollow model, after fixing the file I could indeed see the correct infill in Cura. 

Could you please try to fix the model and retry? Thanks!

PhilmacFLy, 2019-01-30:
I actually got the finished file wiht 12 rupees from https://www.thingiverse.com/thing:2642328 and created the single rupee via shell seperation in Autodesk Netfabb.

But yeah after reparing the 12 Rupees file in Autodesk Netfabb and loading it into Cura again, I also get infill for it. So I guess its just that the file is borken

closed 2019-01-30 by PhilmacFLy