Ultimaker/Cura #5247: [Cura 4.0] Slice Hangs Forever - Trying to retrieve setting with no value given: 'minimum_support_area'

Enverex, 2019-01-30:
**Application Version**
4.0 Beta

**Platform**
Linux x86_64

**Printer**
Wanhao Duplicator 6

**Steps to Reproduce**
Set printer to Wanhao Duplicator 6 profile, import any models (seems to happen with anything), then try and slice.

**Actual Results**
Slicing bar gets about 1/3 the way across and stops. It doesn't appear to ever progress past this point.

**Expected results**
For slicing to complete.

**Additional Information**
Command line shows that it appears to be erroring on a missing variable, specifically **[Backend] [ERROR] Trying to retrieve setting with no value given: 'minimum_support_area'**. The full output from the slicing process was:

```
2019-01-30 14:58:45,724 - DEBUG - [Thread-11] UM.Backend.Backend._backendLog [101]: [Backend] [WARNING]  -s initial_extruder_nr="0" -s day="Wed" -s time="14:58:44" -s print_temperature="210" -s print_bed_temperature="40.0" -s cool_fan_full_at_height="0.3" -s infill_extruder_nr="-1" -s default_material_bed_temperature="40.0" -s wireframe_printspeed_down="5" -s bridge_wall_coast="100" -s smooth_spiralized_contours="True" -s skin_no_small_gaps_heuristic="False" -s retraction_speed="25" -s meshfix_maximum_travel_resolution="0.02" -s machine_nozzle_heat_up_speed="2.0" -s spaghetti_max_height="2.0" -s retraction_combing_max_distance="0" -s jerk_ironing="20" -s prime_tower_line_width="0.4" -s acceleration_wall_0="3000" -s bridge_fan_speed_2="0" -s machine_max_jerk_e="5.0" -s connect_infill_polygons="False" -s center_object="False" -s wall_min_flow="0" -s travel_avoid_other_parts="True" -s filter_out_tiny_gaps="True" -s speed_slowdown_layers="2" -s skirt_line_count="1" -s infill_pattern="grid" -s jerk_travel="30" -s wireframe_printspeed_flat="5" -s cooling="0" -s speed_support_infill="60" -s infill_multiplier="1" -s default_material_print_temperature="210" -s retraction_hop_only_when_collides="False" -s support_tree_angle="40" -s platform_adhesion="0" -s acceleration_support_interface="3000" -s machine_show_variants="False" -s prime_tower_circular="True" -s support_bottom_line_width="0.4" -s infill_overlap_mm="0.04" -s skin_angles="[ ]" -s support_bottom_enable="False" -s cool_fan_speed_min="100.0" -s speed_wall_x="60.0" -s layer_height="0.1" -s retraction_retract_speed="25" -s support_top_distance="0.1" -s material_adhesion_tendency="10" -s raft_surface_line_spacing="0.4" -s acceleration_travel_layer_0="5000.0" -s machine_width="200" -s z_seam_relative="False" -s machine_steps_per_mm_x="50" -s retraction_combing="all" -s bridge_skin_speed_3="15.0" -s bottom_layers="8" -s blackmagic="0" -s infill_line_distance="4.0" -s switch_extruder_prime_speed="20" -s sub_div_rad_add="0.4" -s mold_angle="40" -s machine_max_jerk_xy="20.0" -s jerk_topbottom="20" -s spaghetti_infill_enabled="False" -s support_pattern="zigzag" -s machine_name="Wanhao Duplicator 6" -s layer_start_x="0.0" -s speed_wall="30.0" -s prime_tower_min_volume="6" -s bridge_skin_speed_2="15.0" -s gradual_support_infill_step_height="1" -s bridge_skin_density_3="80" -s wireframe_top_delay="0" -s speed_equalize_flow_enabled="False" -s cool_fan_speed_0="0" -s date="30-01-2019" -s machine_heat_zone_length="16" -s material_initial_print_temperature="200" -s travel_compensate_overlapping_walls_enabled="True" -s gradual_infill_steps="0" -s z_seam_y="600" -s support_join_distance="2.0" -s fill_outline_gaps="False" -s raft_surface_line_width="0.4" -s draft_shield_height="10" -s adhesion_extruder_nr="0" -s relative_extrusion="False" -s speed_equalize_flow_max="150" -s support_infill_extruder_nr="0" -s wall_line_count="2" -s bridge_wall_material_flow="50" -s wireframe_strategy="compensate" -s coasting_speed="90" -s support="0" -s spaghetti_inset="0.2" -s mold_roof_height="0.5" -s resolution="0" -s prime_tower_position_y="170.4" -s bridge_wall_min_length="5" -s alternate_extra_perimeter="False" -s top_bottom_thickness="0.8" -s raft_surface_thickness="0.1" -s conical_overhang_enabled="False" -s jerk_travel_layer_0="30.0" -s bottom_thickness="0.8" -s raft_surface_jerk="20" -s travel="0" -s initial_layer_line_width_factor="100.0" -s raft_interface_jerk="20" -s support_interface_height="1" -s roofing_pattern="lines" -s jerk_print_layer_0="20" -s support_enable="False" -s line_width="0.4" -s wireframe_drag_along="0.6" -s anti_overhang_mesh="False" -s top_bottom_extruder_nr="-1" -s machine_max_acceleration_y="9000" -s support_bottom_distance="0.1" -s acceleration_wall="3000" -s wireframe_roof_outer_delay="0.2" -s machine_nozzle_head_distance="3" -s skirt_brim_line_width="0.4" -s speed_support_bottom="40.0" -s slicing_tolerance="middle" -s magic_fuzzy_skin_enabled="False" -s wireframe_flat_delay="0.1" -s wireframe_flow="100" -s meshfix_keep_open_polygons="False" -s ironing_pattern="zigzag" -s wireframe_roof_fall_down="2" -s retraction_extra_prime_amount="0" -s material_flow_layer_0="100" -s start_layers_at_same_position="False" -s acceleration_wall_x="3000" -s raft_interface_acceleration="3000" -s ooze_shield_angle="60" -s extruder_prime_pos_abs="False" -s speed_ironing="20.0" -s support_extruder_nr="0" -s prime_blob_enable="True" -s prime_tower_enable="False" -s raft_interface_speed="22.5" -s wall_line_width_0="0.4" -s material_bed_temp_prepend="True" -s bridge_skin_support_threshold="50" -s support_zag_skip_count="8" -s skirt_gap="3" -s ironing_enabled="False" -s infill="0" -s raft_interface_line_spacing="1.0" -s wall_0_extruder_nr="-1" -s machine_steps_per_mm_e="1600" -s retraction_min_travel="0.8" -s gradual_support_infill_steps="0" -s machine_nozzle_size="0.4" -s brim_line_count="20" -s raft_base_speed="22.5" -s material_surface_energy="100" -s wireframe_height="3" -s ooze_shield_dist="2" -s draft_shield_enabled="False" -s adaptive_layer_height_enabled="False" -s brim_width="8.0" -s machine_endstop_positive_direction_x="False" -s machine_use_extruder_offset_to_offset_coords="True" -s acceleration_support_infill="3000" -s jerk_skirt_brim="20" -s speed_roofing="30.0" -s support_roof_line_width="0.4" -s optimize_wall_printing_order="False" -s machine_endstop_positive_direction_z="True" -s mold_width="5" -s bridge_skin_density="100" -s speed_prime_tower="60" -s mesh_position_z="0" -s meshfix="0" -s speed_wall_0="30.0" -s support_skip_zag_per_mm="20" -s alternate_carve_order="True" -s support_initial_layer_line_distance="2.6666666666666665" -s ironing_flow="10.0" -s support_extruder_nr_layer_0="0" -s support_roof_extruder_nr="0" -s cool_min_layer_time_fan_speed_max="10" -s jerk_layer_0="20" -s cool_fan_speed_max="100.0" -s roofing_extruder_nr="-1" -s support_xy_distance="0.7" -s cool_fan_enabled="True" -s speed_travel="120" -s wireframe_up_half_speed="0.3" -s machine_extruder_count="1" -s top_thickness="0.8" -s minimum_polygon_circumference="1.0" -s skirt_brim_speed="30.0" -s layer_height_0="0.3" -s bridge_fan_speed="100" -s ooze_shield_enabled="False" -s machine_center_is_zero="False" -s cool_min_layer_time="5" -s acceleration_infill="3000" -s speed_topbottom="30.0" -s raft_base_line_width="0.8" -s jerk_support_bottom="20" -s ironing_line_spacing="0.1" -s material_final_print_temperature="195" -s raft_jerk="20" -s coasting_enable="False" -s support_infill_sparse_thickness="0.1" -s acceleration_topbottom="3000" -s skin_outline_count="1" -s support_conical_angle="30" -s material_shrinkage_percentage="0" -s skin_line_width="0.4" -s adaptive_layer_height_variation="0.1" -s flow_rate_extrusion_offset_factor="100" -s raft_base_jerk="20" -s experimental="0" -s machine_head_with_fans_polygon="[[-20, 10], [10, 10], [10, -10], [-20, -10]]" -s wall_extruder_nr="-1" -s support_type="everywhere" -s wireframe_roof_drag_along="0.8" -s jerk_support_roof="20" -s jerk_roofing="20" -s raft_surface_acceleration="3000" -s bridge_skin_speed="15.0" -s max_feedrate_z_override="0" -s adaptive_layer_height_variation_step="0.01" -s support_bottom_density="100" -s machine_feeder_wheel_diameter="10.0" -s support_angle="50" -s infill_wipe_dist="0.1" -s travel_avoid_supports="False" -s shell="0" -s infill_wall_line_count="0" -s speed="0" -s speed_print="60" -s brim_replaces_support="True" -s jerk_support="20" -s draft_shield_dist="10" -s acceleration_print_layer_0="3000" -s limit_support_retractions="True" -s acceleration_support="3000" -s xy_offset="0" -s roofing_layer_count="0" -s material_bed_temperature_layer_0="40.0" -s acceleration_support_roof="3000" -s machine_heated_bed="True" -s magic_fuzzy_skin_point_density="1.25" -s support_bottom_stair_step_width="5.0" -s wireframe_top_jump="0 [... cut ...]
2019-01-30 14:58:48,249 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [217]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2019-01-30 14:58:48,249 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [197]: Socket state changed to Closed
2019-01-30 14:58:48,273 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [810]: Backend quit with return code 2. Resetting process and socket.
2019-01-30 14:58:53,527 - DEBUG - [MainThread] cura.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
```

Ghostkeeper, 2019-01-31:
I can't reproduce it here. Not using the current 4.0 state nor using the public beta. It slices fine for the Wanhao Duplicator 6. Also with support enabled.

Also looking into the code, the `minimum_support_area` setting is present.

Enverex, 2019-01-31:
May have been an issue with the 4.0 branch at the time. I've just rebuilt from master instead and it's working fine. Thanks for checking, looks like it's a non-issue now.

closed 2019-01-31 by Enverex