Ultimaker/Cura #5248: Limit extruder feedrate for printing and retraction

MarkMan0, 2019-01-30:
**Application Version**
3.6.0

**Platform**
Windows 10

**Printer**
With Marlin firmware

So I'd like to have a feature to limit the maximum speed of the extruder, while printing.
I've tried doing this by chaging the limits in firmware, but it made retraction slow too, obviously.

So what I imagine is a new setting in the speed tab, like retraction speed
Extruder max. speed [mm/s]   - lets say 3mm/s. Thats 180 mm/min

And then in the gcode it would be like:
.....
M203 E180 ; set extruder speed for printing
; printing now
G1
G1
G1
...
;Now retract, reset feedrate
M203 E...
G1 ;retract
G0 ;travel
G0
G1 ; un-retract
M203 E180  ; reset feedrate

**Why I think this is good?**
We can easily measure the max. feedrate our printer can handle with certain filaments. For me it's 200mm/min. This value doesn't change for different layer heights, so no need to play with temperature/speed settings when increasing layer height.

The max print speed would be limited in firmware.

Ghostkeeper, 2019-01-31:
This has been asked of us before but I can't find the original back right now.

We would prefer to implement this in a somewhat different way: Using inheritance formulas in the front-end. You would set the maximum flow rate as a setting in mm³/s, and the setting for the print speed would be a formula such as this:

    speed_print = =max_flow / layer_height / line_width

And then the printing speed would be adjusted down if you increase the layer height.

What I'm not really sure about yet though is how that would look if there are also other constraints to the print speed, such as accelerations in the outer wall causing ringing, steppers losing steps when going too fast, etc.

MarkMan0, 2019-02-02:
Sounds great! Looking forward to it :)

At first sight it doesn't look like it's going to cause ringing, you're essentially just adjusting the max print speed.
The final formula for the speed could be something like:
`speed_print = min( (max_flow / layer_height / line_width), user_set_max_speed)`

I would only use it to insure, that I'm not printing faster than my extruder can handle. 
I got the idea when printing 0.32 infill and 0.16 wall, at 100mm/s. The walls were fine, but the infill was under-extruded.

For now I'll write myself a parser

MrStump, 2019-09-09:
> This has been asked of us before but I can't find the original back right now.
>
> We would prefer to implement this in a somewhat different way: Using inheritance formulas in the front-end. You would [...]

Hi, any news on this? this really helpfull feature

Ghostkeeper, 2019-09-09:
Not really, but it depends on the printer you're using. Realistically it's hard to get this sort of thing through for Ultimaker printers, since all of Ultimaker's profiles need to be re-tested then. For other printers we're dependent on people from the community that have those printers to test, but since they don't have a setting to test with it's not going to be easy for them. A bit of a circle of dependencies going on.

MrStump, 2019-09-09:
> Not really, but it depends on the printer you're using. Realistically it's hard to get this sort of thing through for Ultimaker printers, since all of Ultimaker's profiles need to be re-tested then. [...]

may be just add setting for "extrusion speed" in the speed tab, or material tab? and user can calibrate it themself. it's like calibrate step/mm for extruder but trough hot nozzle, command "feed" 100mm and measure is it really go 100mm through extruder or just 85mm.
for ex. my extruder can melt plastic with 3.5mm/sec direct flow. with 0.8 nozzle and 0.4layer it can provide max feedrate is ~70-80mm/sec, with 0.3 layer it can 80-100mm/sec. for this moment i test it with every nozzle/layer change, but if we'll have setting like this then slicer will limit max print speed depends on "raw extrusion speed", not "feeder speed". for now it could be added in "experimental tab". really, it would be reeeealy helpful.

Ghostkeeper, 2019-09-09:
If the "extrusion speed" changes, would it change the line width or the printing speed?

MrStump, 2019-09-10:
It should limit printing speed. Because if I want smaller line width i can set it. But if extruder can't provide needful amount of plastic on current printing speed with current line width and layer height then printer should slowdown. 
For this moment with plugin "printer settings" i can set "maximum feedrate" but it limit whole extruder speed and i can't use retract.

Ghostkeeper, 2019-09-10:
Hmm, I think the most major concern peft then is that the printing speed settings won't work any more then. Either that, or the extrusion speed setting won't work if the printing speed settings are overridden by any profile.

MrStump, 2019-09-10:
in my opinion "extrusion speed" should be in charge, by default it could be as lightspeed, but if user want to calibrate "melt plastic extrusion speed" then feedrate should be limited by it. For example i don't really know is my priter able melt "this plastic" through 0.8 nozzle with layer 0.4 at speed 100mm/sec. but if there will be an option "melt plastic extrusion speed" that i checked before (like 3mm/sec) then slicer limit feedrate speed to 92mm/sec, because slicer can calculate what really speed of "melt plastic extrusion speed" will be in that setup and i'm not :) 
i mean if through 0.8 nozzle with layer 0.4 at speed 100mm/sec my extruder should feed raw plastic with 3.5mm/sec - my extruder would'n able to do it right, gear start to slip on raw plastic, there is clogging in nozzle and print will not be successfull. in case when we have THAT option CURA will show me that speed of infill with 100mm/sec is "yellow warning" or just limit it when sliced.
sorry for my not realy good english.
Regards,
Stump )

Ghostkeeper, 2019-09-11:
You mean that there should be a "maximum extruder feed rate" setting, then, but one that does not limit the retraction speeds as well.

MrStump, 2019-09-11:
yes. actually i don't know when CURA calculate actual speed of extruder feedrate, i think CURA do it only when it slicing.
so it should limit whole printing speed limited with "maximum extruder feed rate" while it calculate all. by default it could be at "light speed"
it's like "Minimum Layer Time", but vice versa

perfessor101, 2020-03-15:
I would really like this feature too, a setting for the Maximum Volumetric Flow rate of the Extruder HotEnd combination.
Quite often I'll set my speeds thinking I have things dialed in for that print and then ninety minutes into the print the extruder will start clicking causing under extrusion as it snaps back and possibly other imperfections as it snaps forward again. 
At that point I have to decide if I would rather have the extra imperfections or try again throwing out what has been printed so far.

CNC Kitchen has a video showing how at higher volumetric flows the amount extruded drops off and can be somewhat compensated for with temperature increases. This could help with keeping those volumetric flowrate inconsistencies in check .

dariuskt, 2020-03-29:
That is a very needed feature for me too. With 0.8 and 1.0mm nozzles extruder becomes the only limit for print speed.

Looks like this feed limit should naturally behave like Minimum Layer Time. It is a dynamic limitation as Layer Height and Line Width may be variable through the print.

Units for this configuration aren't that important as mm³/s and mm/s are basically the same and cant change independently for a given filament width.

sbstnzmr, 2020-06-06:
Hey. I would really love to see that feature too. Not only for the larger nozzle diameters but also for filaments like TPU this is really useful.

As far as my experiences with other slicers go I would assume an implementation liket this: For PLA afaik 12-15 mm³/s is a good starting point for a regular hotend. This should limit the maximum printing speed. Or not if you assume a 0.4 nozzle because your printer can probably not go that fast so your regular printing speed applies. It's basically a max function on your volumetric speed and your print speed settings. For a 0.8 nozzle and a 0.4 layer height it would automatically calculate the maximum speed and still respect your regular speed settings otherwise. For a volcano it's more in the 30 mm³/s department so I personally don't really need it however it does make a difference for common hotends.

For TPU/Flex this makes adjusting settings also much easier. You could just set your max volumetric speed to something like 1.5 mm³/s and assuming a .2 layer height and a .44 line width your print speed would cap at 1.5/(0.2*0.44). Like this I can basically print speed constrained materials without messing with print speed settings.

I know what to set manually but especially more inexperienced users who would try larger nozzles on a common hotend would benefit from not messing up the prints and clogging the hotend. And I just miss this setting since it seems like a calculation that could be possible.

perfessor101, 2020-06-06:
For an ender 3 with a 0.4mm nozzle it’s closer to 6 or 7 cubic mm/s ... 50 mm/s with 0.5mm nozzle 0.635 mm line width and 0.24 mm layer height clicking away ... dropped down to 45%.  So much easier to limit it by volumetric rate. Then if I make a mistake (Going too fast) I don’t have to slow everything down.

AndrewEllis93, 2020-06-25:
+1. This is the primary thing that stops me from using Cura as much as I'd like.

MrStump, 2020-06-29:
any news about Volumetric speed limits?

Ghostkeeper, 2020-07-02:
Sorry, no, we haven't worked on this yet.

Aggebitter, 2020-07-06:
Some addon to the tread. The Volumetric speed aka mm3/s is critical on all FDM printers! If you hit that limit under extrusion  is a fact. 

I had an issue with my bowden tube and made some tests and calculations (good page for the math and how to test http://projects.ttlexceeded.com/3dprinting_calibrating_volumetric_rate.html)

My results with a stock Volcano head 0.8mm nozzle on stock Flsun Q5 gave me this:

Filament: PrimaValue Black PLA

(bad bowden tube)
Temp		Max Extrusion speed	mm3/s (F/60 * A (2.403 for 1.75) )
M109 S200	G1 E100 F200		8.01 mm3/s	
M109 S210	G1 E100 F350		12.015 mm3/s
M109 S220	G1 E100 F500		20.025 mm3/s

(changed to unbranded one laying around)
M109 S190	G1 E100 F550		22.0275 mm3/s
M109 S200	G1 E100 F600		24.03 mm3/s
M109 S210	G1 E100 F650		26.0325 mm3/s
M109 S220	G1 E100 F750		30.025 mm3/s 

30mm/s is the max my setup can extrude in open air. That limit should never be passed. Adding a safety margin for back pressure when extruding will have to be applied but in my case that was needed when I printed at 0.10mm layer with a 0.8mm nozzle (yes it did print just fine at 1mm line width) 

To control extrusion with mm/s instead of mm2/s gives you a lot of errors you have to tweek. 
-The gripping diameter of the filament pulley is depending on how soft the filament is. The gripping radius will change and thereby the distance per step will change. 
-The nozzle pressure depends a lot on nozzle temperature for each filament used
-The layer height generates different back pressure especially when using "ironing" with a wider line width than nozzle diameter

By using mm2/s we will reduce the needed tweaking to just for each filament to for constants:

Temperature: C
Layer height: mm
Line width: mm 
Volumetric: mm3/s

And we will get max printing speed in mm/s by doing the math
20mm3/s @ 0.2mm layer and 1mm line width gives:
20mm3 / (0.2mm * 1mm) = 100mm/s max speed

After that calculation we can start doing all the nice printout tweaks
//Agge

Glodigit, 2020-12-19:
+1 on this

sgessner, 2021-01-09:
+1 ; Would be nice to have that feature!

controlflow, 2021-01-16:
+1, I really miss this option

Jackjan4, 2021-01-25:
+1 however I would implement it as maximum volumetric flow rate since PrusaSlicer has it and its fantastic for working out what the extruder is capable to do with different filaments.

rcmonitor, 2021-02-14:
+1 for mm/time unit setting (mm/s or mm/min).
Volume of material may vary (the filament spool may be of let's say 2.90 instead of 2.85, or 1.70 instead of 1.75), but the performance of extruder on filament may not.

conceptual-perception, 2021-03-26:
I have a Prusa M3KS and ... changing out the nozzle from 0.4 to 0.8 ... broke Cura completely. I started a print and ... my Octoprint setup threw a M112 message. It took me a bit to understand what had happened, that Cura instructed the printer to send filament faster then the printer could safely extrude it. Thankfully, my printed detected this as a thermal runaway event and forced a reset.

I don't really see from a user perspective how this is possible, since the MVS value completely prohibits this. My E3D v6 hotend can only handle 15mm^3 a second. Where do I put this value into Cura? I thought adjusting the nozzle size would ... change other settings (it didn't), but ... idk if it's worth me going in and figuring out all the speeds for all the parts of the print. That seems like work software should do.

https://help.prusa3d.com/en/article/max-volumetric-speed_127176

```
The Maximum volumetric speed setting (MVS) is one of the most powerful features in PrusaSlicer. The MVS setting essentially creates a manager for the maximum amount of filament that the slicer will attempt to push through your 3D printer’s hotend.

This gives PrusaSlicer users a huge advantage. Rather than trying to calculate a range of speeds for combinations of different nozzle sizes, layer heights or extrusion widths, you can simply specify the speeds and settings you desire, then let MVS regulate speeds at slice time when and only if necessary.
```

I love Cura, but this missing, hurts.

pemerick07, 2021-04-06:
+1 For this. I like Cura much better than Slic3r/PrusaSlicer, but I always miss this feature. It's also incorporated into Fusion 360 Additive as a maximum E feedrate for given hardware configuration.

beamzer, 2021-04-15:
+1 for this, i came here looking for how to adjust this in cura.

perfessor101, 2021-05-10:
> So what is the current workaround for this missing basic feature?

What I do is start a print … wait until the extruded starts clicking …throw that one out and adjust speeds down … start another print  …. Wait until the extruded starts clicking and throw that out …

So far that’s what we do with Ultimaker cura.

conceptual-perception, 2021-05-10:
I've been manually figuring out the speeds for my prints. I switched to a
1mm nozzle, and I went through and made sure the height x speed x nozzle
isn't faster than the MVS.

My hotend can do 11.5 cm^3/sec so the maximum speed at a layer height of
0.2 is about 57.5cm/s.

perfessor101, 2021-05-11:
I use adaptive layer heights … so I guess it’s one or the other.

Ghostkeeper, 2021-05-18:
Cura 4.10 will contain a visualisation for the flow rate. It's not automatically adjusting settings to limit the flow rate, but it at least allows visualisation of it. See https://github.com/Ultimaker/Cura/pull/9410.

fvrmr, 2021-06-01:
We have created a ticket for adjusting settings to limit the flow rate on our backlog.
Devs se CURA-8282

tastyratz, 2021-06-01:
There are per printer flow limits, but, they differ per filament. 
Think about all the weird changes you do black magic calculations around that could otherwise be measured and calculated.

Peak extruder flow for a printer as a whole. Peak flow for a filament based on temperature, we could even do a flow rate calculation 2d mapping against temperature (weigh open air output at xmm/s at xdeg, then again at higher speed or temp. It would make a more solid material profile where we wouldn't need to mess with extrusion multipliers or temps to indirectly change flow.

Earlier in this req mentions Marlin codes, but, I'm running Klipper for the advanced featureset and resonance compensation. This should be firmware agnostic.

AbeFM, 2021-09-20:
I've seen the limit set per filament in the past, this can make a lot of sense... Although really it is a combination of material and heater block....
I was just checking the thread to request at least a visulalization, I'd discounted "flow" as being a percentage (flow/wall flow/prime tower flow).... Perhaps **change "Flow" to "Flow Rate"**?

Glad to see this is getting attention.

rostislav-marinov, 2021-09-22:
If we remove some other factors and take only flow speed to be the only limit and have to be thinner layers prints faster than thick layers and slimmer line prints faster than thicker line. Than adaptive layers will print faster. Then you will have speed changing not only for features of the objects, but the layers it's self. Also you will not sit and calculate every speed for every layer hight/thickness  combination. 

I'm wondering... Maybe I can try to make a plugin.
So you can set 8 mm³/s, as main flow, 4mm³/s for outer, 50% for overhangs, etc.

Glodigit, 2021-09-22:
Is there a way to get Cura to compute the values by modifying the filament profile manually? Like print_speed = [max speed] or [max flow rate speed calculation]. I'd totally open up VS code for 30 mins and now I have a set + forget filament profile.

AbeFM, 2021-09-22:
It seems a maximum extruder speed would work? I'm unclear what happens if you set a high speed line and are against a firmware speed limit - will the whole move slow down, or will you underextrude but move the printhead at the same rate?

When I look at code coming out of PrusaSlicer, it is aware of hitting that limit and orders slower moves where more material is coming out (in my case, thicker walls). That IS the purpose - to limit volumetric flow rates - which prevents jams and grinding. It is dependent on material (and to a degree, temperature).

It might be hard to implement (though not impossible as other slicers have done this for years). I can't tell if Rostislav's comment, while accurate, is saying we should or shouldn't do this because of x-y speed changes, but I feel we should. 

Alternatively: Could you add a graphical only limit - when displaying the print speed graph, could areas which are over a particular print speed be highlighted as overhangs currently are?

rostislav-marinov, 2021-09-23:
I'm currently testing different approach/workflow.
Running Klipper on Modded Anet ET4x with bad hard to print PLA filament (cheapest locally made) on 0.3mm nozzle. 
- Found my maximum flow rate and let say optimal is 70%-80% from maximum.
- Found my Acceleration Limits.
- Found my Linear/Pressure Advance (Klipper uses this this very good internally).
- Using Google sheet to calculate speeds. https://docs.google.com/spreadsheets/d/1reUdOvrVpZviGmEmnvC4giYMgKwU9U2PYtte8ioz1T4/edit?usp=sharing

Testing out those speeds atm.

unkn0wndude, 2021-09-24:
> I'm currently testing different approach/workflow.
> Running Klipper on Modded Anet ET4x with bad hard to print PLA filament (cheapest locally made) on 0.3mm nozzle.
>
> * Found my maximum flow rate and [...]

Could you add option to set nozzle size?

rostislav-marinov, 2021-09-27:
Yep there is issue as the "safe" margin with the maximum speeds for external walls decreases. If you have like 75% for 100mm/s it should be 90% for 30mm/s or something like that. Also formula is quite simple ATM, sheet two I've made bigger data set and speedline is quite steep at the end, for me it should be a bit more straight so speeds are a bit more conserve and safer.

Ghostkeeper, 2021-09-27:
> It might be hard to implement (though not impossible as other slicers have done this for years). I can't tell if Rostislav's comment, while accurate, is saying we should or shouldn't do this because of x-y speed changes, but I feel we should.

I don't think any of the suggestions here are difficult to implement. The biggest problem is that all of these settings are a bit of a mess to the user and also overconstrained. If the user specifies 50mm/s, 0.2mm layer height and 0.4mm line width they are currently getting 4mm³/s. If the extrusion rate limit is set to 3mm³/s, they'll see that their speed is reduced everywhere and might be wondering why. Cura can't use both the correct speed and the correct extrusion rate. It's going to have to choose one over the other and the user must somehow be intuitively know why.

If we implement this feature via inheritance formulas in the front-end, we can make the speed setting automatically adjust based on the extrusion rate limit, and users may override that if they wish, but it won't work well for cases like Minimum Layer Time where the speed is adjusted in firmware.
If we implement it in CuraEngine we can apply the flow rate limit everywhere correctly, but the user doesn't get any feedback on why it's not printing at the speed set by the speed setting.

Personally I'd think the solution in the front-end would be best. For the Minimum Layer Time you probably want to have a lower maximum extrusion rate anyway to make it cool better. There may be other speed adjustments I'm not thinking about now, but those are details we can print with.

> I'm wondering... Maybe I can try to make a plugin.

If you're attempting this, I'd suggest taking a look at the [Linear Advance Setting Plug-in](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin) to see how fieldOfView added a setting to the list safely.

rostislav-marinov, 2021-09-27:
> > I'm wondering... Maybe I can try to make a plugin.
> 
> If you're attempting this, I'd suggest taking a look at the [Linear Advance Setting Plug-in](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin) to see how fieldOfView added a setting to the list safely.

Thank you for the reference it's a very good starting point, as soon as I have a bit more time will try to make some basic implementation.
From my tests main issues with this approach to speed are cooling/bridging and sticking/overhangs. You can have a bit higher speeds with thinner extrusion, but not as direct increase like with the simple shells and perimeters. Just to point again I'm currently testing with filament that is cheap and unreliable.

Life0fBrian, 2021-10-17:
Where would be the problem to implement that parameter into lets say **Machine Settings** -> **Extruder X** just underneath the **Compatible material diameter**.
The default value shall be 0 and if a user changes this to his determined value maybe a short notice will remind him to the possible speed changes now caused by that limiting factor.

However an experienced user who wants to determine the max volumetric extrusion rate should already know that it affects speed during printing.

And yes there will always be people who just want to have a one button does it all setting/GUI and others who want to be able to change every single parameter

Ghostkeeper, 2021-11-01:
Arachne (in its current state, not yet in the beta from March) has a setting to equalise the material flow by adjusting the movement speed, too. It causes the flow rate to be adjusted to the nominal flow rate for that line type (speed * layer height * line width), so it doesn't solve the original issue here which is about the flow difference between different line types. But it's related.

Arsoth, 2022-02-28:
+1, would love a max flow rate that I can put in. There's a reason it seems like Voron users all run SS/PS, because getting some of the granularity in Cura is a PITA. But Cura's UI is WAY better so... I would love it it had as many features and settings.

psiberfunk, 2022-04-06:
Another pile on user here.  Having built my own printers at this point and used Klipper and Marlin and Cura and PrusaSlicer and SuperSlicer (SuSi), I can definitely say I generally prefer Cura's way of doing things, but SuSi/PR's way of doing volumetric-constrained speed throttling is absolutely a godsend to setting speeds.

In my case, my CoreXY printer is great at handling high speed moves and i'm often more constrained by the filament properties and the extruder's ability to press things through the hotend (Dragon Highflow) than I am by speed limitations per se.  Setting speed limits within the context of a volumetric flow is so much more intuitive when used in combination with speed limits ... here's how I think about it:

Volumetric flow is like my home internet pipe, throttling the total bandwidth to my devices in my house.  You can't exceed it's compacity without upgrading.  Everything must bow to the ISP link speed.  Each of the other speed limits I may set in my router quality-of-service settings are subservient to the master bandwidth limit.  Figuring out what is limiting during the print could easily be messaged to the user with a popup tip on slicing.. or better.. when viewing the speeds/flow views of the slicer preview, where it would be most appropriate.  

I would expect to get a little notice that says "warning, your speeds have been capped by your maximum flow rate (XX) in this print.  Learn more <insert link to page explaining basic concepts>"

khani3s, 2022-04-15:
So basically it's just an UI problem? 
Let's just add an option under experimental for now. If someone enables it just paint all overridden speeds on orange and set a popup message like others suggested.
If someone turned on advanced mode I believe he is already able to get around in the settings.

to-the-nth, 2022-07-17:
I have found that some materials like TPU are volume flow limited.   I was getting Extruder jams on my Voron using SuSi because I was allowing line width to vary for interior fills at "low" print speeds.  There were a few fat lines where the volume spiked resulting in a jam.  Setting the max flow rate worked like a charm. So far, I am reliably able to run TPU at volume flow rates of 6mm^3/sec (runtime confirmed on the mainsail display) which if you do the math for 0.4 wide line and 0.2 layer height yields print speeds of 75mm/sec.  Print speeds are vastly increased.  Reading this thread, it is not clear what requests are opened, closed and linked.  I was having some other issues with SuSi and was looking at Cura, but not having volumetric flow is a non-started for TPU.

Pennerbert, 2022-11-26:
+1
Any news out there?

smartavionics, 2022-11-27:
Easy to do so I have added a Max Filament Flow Rate setting to my Cura variant and it will be available in the next release (4.20.9) (see https://github.com/smartavionics/Cura/releases).

With no flow limiting...
[image]

With flow limited to 7 mm^3/S...
[image]

Arsoth, 2022-11-27:
> Easy to do so I have added a Max Filament Flow Rate setting to my Cura variant and it will be available in the next release (4.20.9) (see https://github.com/smartavionics/Cura/releases).

Any chance you could put a PR in for that... or is it wildly incompatible with Arachne (I'm sure I misspelled that) engine?

smartavionics, 2022-11-27:
> Any chance you could put a PR in for that... or is it wildly incompatible with Arachne (I'm sure I misspelled that) engine?

Sorry, I no longer submit PR's for anything other than fixes for my own bugs. The good news is that the required changes are trivial so anyone who can build Cura 5 from source could submit an equivalent PR with little effort.

psiberfunk, 2022-11-27:
> Sorry, I no longer submit PR's for anything other than fixes for my own bugs. The good news is that the required changes are trivial so anyone who can build Cura 5 from source could submit an [...]

I’m not sure why you would comment on this thread if you’re to dangle a fix but not going to submit a “trivial” PR.  That’s a little rude IMO.

tastyratz, 2022-11-27:
> 

I mean, I get that you have your own fork which gives you a lot more freedom and I suspect there could be politics being in the big sandbox with everyone else, but, I really hope you reconsider this position. Having your own fork is great when you want to take things a new direction and have a different vision. Having your own fork is simpler. It's also great to put your changes in there if they are denied by the main branch because of fundamental differences on the end result.

If you won't even submit a PR to begin with though, Those things don't really apply. 

I'm glad you made the change and that you feel it's rather trivial to implement Please consider helping everyone at large who isn't using a spinoff.

psiberfunk, 2022-11-27:
> I'm glad you made the change and that you feel it's rather trivial to implement Please consider helping everyone at large who isn't using a spinoff.

Or that doesn't have the deep familiarity with Cura or the build system...

psiberfunk, 2022-11-27:
So, I was just browsing his relevant changes, which I found buried in his branch here (he did not link to them easily for some reason):  https://github.com/smartavionics/Cura/commit/81a6fddfd1d886fe7d69d5ff2d53b787946cbf89?diff=unified

While I don't understand the Cura codebase at all.. it seems like based on the changes this is a slicer front-end setting in config files to somehow trigger the use of a CuraEngine feature that's buried... but I can't find any reference to it in the CuraEngine code.. so i'm lost.

psiberfunk, 2022-11-27:
If someone wanted to pull this in, it would seem you'd need to pull in his additions to the CuraEngine project as well, which I located here:  
[image]
https://github.com/smartavionics/CuraEngine/commits/mb-testing

olmedocr, 2023-01-16:
@psiberfunk are those two changes (engine + cura) the only ones needed? Why don't you try and submit the PR?

maruhe, 2023-01-26:
I use adaptive layer high and on 0,3mm layer and 120mm/s I'm at the end of my revo 
so to print support and infill a little slower for this layers would help and
limiting extruder feedrate / volumetric flow would solve this :-)
as I already see the flow in cura preview i hope there well be a way

mjc-506, 2023-02-01:
Submitted a couple of PRs for this (one for Cura https://github.com/Ultimaker/Cura/pull/14452 , one for CuraEngine https://github.com/Ultimaker/CuraEngine/pull/1814 ) that implement this, copying the code used for 4.20 above https://github.com/smartavionics/Cura/commit/81a6fddfd1d886fe7d69d5ff2d53b787946cbf89?diff=unified 

This works-on-my-machine(tm) (ie, keeps flow rate at or below the limit set) but it does seem simple enough so can't foresee any problems...

[16 more replies]