Ultimaker/Cura #5249: Cura [4.0.0] Mac Printer settings disappear

masc2279, 2019-01-30:
Printer Settings keep disappearing after a few uses. To change nozzle size you must delete printer and re install to get the dialog where you can change your settings.

Working Screen Image ....
[image]

When it fails to work screen image .....
[image]

This is not project specific 

Thank you for using Cura!
-->

**Application Version**
[4.0.0]

**Platform**
MAC

**Printer**
Ender 3

**Steps to Reproduce**
Install printer ... use cura as you  normally would quit  and start slice a few project and at some  point the settings are no longer accessible

**Actual Results**
Cannot change any printer settings like nozzle ...

**Expected results**
I should  be able to  change nozzle sizes without having to un install printer and re install.

nallath, 2019-01-31:
I think this issue has already been fixed by us, but in the meantime (as a work around) you can just change the line width instead of changing the nozzle size.

masc2279, 2019-01-31:
its still in the current public beta so not sure what you mean you all fixed it ... unless its on a non  released version

nallath, 2019-01-31:
The current publicly released beta is from a while ago, so there are quite a few changes & fixes that have not been released yet. 
I've been unable to reproduce it running that version, but it could always be that there is something system specific that's causing the issue.

closed 2020-11-26 by nallath